dhewg/esphome-miot #2: Heartbeat SIID/PIID not really optional

cristianchelu, 2024-02-01:
I'm trying to create a config for [Smart Pet Food Feeder (mmgg.feeder.fi1)](https://home.miot-spec.com/spec/mmgg.feeder.fi1) which doesn't need a heartbeat SIID/PIID and discovered that even though `miot_heartbeat_siid` and `miot_heartbeat_piid` are defined as Optional in the config validator schema, they don't have a default set in `__main__.py`.

With the following yaml I get no editor warnings but the below error on compilation.
```yaml
external_components:
  source: github://dhewg/esphome-miot@main
# ...
miot:
  id: miot_main
# ...
```
```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/pet-feeder.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 407, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/config/.esphome/external_components/b0e0b644/components/miot/__init__.py", line 34, in to_code
    cg.add(var.set_heartbeat_config(config[CONF_MIOT_HEARTBEAT_SIID], config[CONF_MIOT_HEARTBEAT_PIID]))
                                    ~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'miot_heartbeat_siid'
```

This yaml snippet works as intended though and no heartbeat message is being sent to the MCU.
```yaml
# ...
miot:
  id: miot_main
  miot_heartbeat_siid: 0
  miot_heartbeat_piid: 0
# ...
```

ESPHome 2023.12.9
esphome-miot@d64d16b

P.S. Thanks for this repo! I already have some parts of the config working beatifully for my device.

dhewg, 2024-02-01:
How about now?
Let me know if it works!

closed 2024-02-01 by dhewg

cristianchelu, 2024-02-01:
Alright, this was a surprisingly fast fix.
Supplying either one or the other results in a linter error.
Supplying none works the same as supplying both with value 0 -- no heartbeat messages in the logs with `level=DEBUG` after a few minutes.

I have a few more issues I found / feature requests, hope you don't mind the spam.
I thought to try PRs instead, but I believe it's best I leave those for configs :)

Thanks!

commit 5bff0dcc (2024-02-01): fix compilation without a configured heartbeat

Fixes #2.