dhewg/esphome-miot #3: Support switch and binary_sensor as int

cristianchelu, 2024-02-01:
The specs for [Smart Pet Food Feeder (mmgg.feeder.fi1)](https://home.miot-spec.com/s/mmgg.feeder.fi1) use a `uint8 0 ~ 1` value instead of a `boolean true ~ false` to read or control the status of some properties, for example `SIID 6 PIID 1 (Child Lock)`, `SIID 3 PIID 1 (Indicator Lights)`,  `SIID 4 PIID 8 (Food outlet door)` and `SIID 4 PIID 9 (Food storage cover)`.

I'm not really sure if this should be handled as a configuration option in this component or if it's better to leave it as a combination of filters in the esphome config (or I missed some obvious solution, in which case I'm sorry for the time waste!).

I have the following configuration:
```yaml
binary_sensor:
  - platform: "miot"
    miot_siid: 4
    miot_piid: 9
    name: "Top Lid"
    device_class: opening
  - platform: "miot"
    miot_siid: 4
    miot_piid: 8
    name: "Food door"
    device_class: opening
    
switch:
  - platform: "miot"
    miot_siid: 3
    miot_piid: 1
    name: "Indicator lights"
    icon: mdi:lightbulb
    entity_category: config
  - platform: "miot"
    miot_siid: 6
    miot_piid: 1
    name: "Child Lock"
    icon: mdi:lock
    inverted: yes
    entity_category: config
```

For the Child lock switch, here's a verbose log for triggering it via the mcu_command service with `command: "set_properties 6 1 0"`
```
[19:14:19][D][miot:125]: Queuing MCU command 'set_properties 6 1 0'
[19:14:19][V][miot:091]: Received MCU message 'get_down'
[19:14:19][V][miot:174]: Sending reply 'down set_properties 6 1 0' to MCU
[19:14:19][V][miot:091]: Received MCU message 'result 6 1 0'
[19:14:19][V][miot:174]: Sending reply 'ok' to MCU
[19:14:19][V][miot:091]: Received MCU message 'properties_changed 6 1 0'
[19:14:19][V][miot.switch:011]: MCU reported switch 6:1 is: OFF
[19:14:19][V][miot:174]: Sending reply 'ok' to MCU
```
The inverted switch is now on in the UI, and toggling it back off results in the following verbose log
```
[19:16:18][D][switch:016]: 'Child Lock' Turning OFF.
[19:16:18][V][miot.switch:017]: Setting switch 6:1 ON
[19:16:18][D][miot:125]: Queuing MCU command 'set_properties 6 1 true'
[19:16:18][D][switch:055]: 'Child Lock': Sending state OFF
[19:16:18][V][miot:091]: Received MCU message 'get_down'
[19:16:18][V][miot:174]: Sending reply 'down set_properties 6 1 true' to MCU
[19:16:19][V][miot:091]: Received MCU message 'result 6 1 0'
[19:16:19][V][miot:174]: Sending reply 'ok' to MCU
[19:16:19][V][miot:091]: Received MCU message 'properties_changed 6 1 0'
[19:16:19][V][miot.switch:011]: MCU reported switch 6:1 is: OFF
[19:16:19][D][switch:055]: 'Child Lock': Sending state ON
[19:16:19][V][miot:174]: Sending reply 'ok' to MCU
```

For the read-only top lid state, here's the verbose log when I physically open it:
```
[19:18:33][V][miot:091]: Received MCU message 'properties_changed 4 8 1'
[19:18:33][V][miot.binary_sensor:011]: MCU reported sensor 4:8 is: OFF
[19:18:33][V][miot:174]: Sending reply 'ok' to MCU
```
and when I physically close it back:
```
[19:19:35][V][miot:091]: Received MCU message 'properties_changed 4 8 0'
[19:19:35][V][miot.binary_sensor:011]: MCU reported sensor 4:8 is: OFF
[19:19:35][V][miot:174]: Sending reply 'ok' to MCU
```

ESPHome 2023.12.9
esphome-miot@5bff0dc

dhewg, 2024-02-01:
How annoying... I'll see how we can support that without making the code too ugly :)

dhewg, 2024-02-02:
Can you try with that ^ commit?
You can then use e.g.:
```
binary_sensor:
  - platform: "miot"
    miot_siid: 4
    miot_piid: 9
    miot_true: "1"
    miot_false: "0"
    name: "Top Lid"
    device_class: opening
```

(and the same for switches)

closed 2024-02-02 by dhewg

dhewg, 2024-02-02:
Nice, thanks for the confirmation!

commit 90d95bbb (2024-02-02): allow custom boolean values

While the MIoT specs define the 'bool' type with values 'true' and
'false', some devices apparently use an 'int' with '0' and '1'
instead...

Don't force the 'bool' type for switches and binary sensors, instead
allow the user to set the values representing the boolean true and false
state using the new 'miot_true' and 'miot_false' config variables.

They're optional and default to 'true' and 'false' to cover the sane
MIoT 'bool' type.

Fixes #3.