dhewg/esphome-miot #4: Support `event_occurred` MCU message

cristianchelu, 2024-02-01:
For the [Smart Pet Food Feeder (mmgg.feeder.fi1)](https://home.miot-spec.com/s/mmgg.feeder.fi1), there's a physical/manual feed button that when pressed generates a number of messages from the MCU in the form `event_occured 4 3 5 255`.

This seems to conform to the message format that I read in https://blakadder.github.io/miot/#event_occured , even though the device spec page doesn't list any events.

For this device, I haven't found a cumulative "portions dispensed today/total"property  so counting the portions as they're dispensed is useful for deriving state.

I'm not sure if this is already considered in the TODO comment,
```
 * add "access" to components? read/write/notify
 * automations?
 ```
But leaving this here just in case it's useful.

----

I would deduce the below event stream to be 
1. feed started + id 255 (manual instead of scheduled?)
2. feed ended + dispensed portion count + id 255 (manual instead of scheduled?)
3. ??

Example short press.
```
[20:25:21][V][miot:091]: Received MCU message 'event_occured 4 3 5 255'
[20:25:21][W][miot:297]: Unknown command 'event_occured'
[20:25:21][V][miot:174]: Sending reply 'ok' to MCU
[20:25:21][V][miot:091]: Received MCU message 'net'
[20:25:21][V][miot:174]: Sending reply 'cloud' to MCU
[20:25:21][V][miot:091]: Received MCU message 'get_down'
...
[20:25:24][V][miot:174]: Sending reply 'down none' to MCU
[20:25:24][V][miot:091]: Received MCU message 'event_occured 4 2 4 1 5 255'
[20:25:24][W][miot:297]: Unknown command 'event_occured'
[20:25:24][V][miot:174]: Sending reply 'ok' to MCU
[20:25:24][V][miot:091]: Received MCU message 'get_down'
[20:25:24][V][miot:174]: Sending reply 'down none' to MCU
[20:25:24][V][miot:091]: Received MCU message 'event_occured 4 1 4 1 5 255'
[20:25:24][W][miot:297]: Unknown command 'event_occured'
[20:25:24][V][miot:174]: Sending reply 'ok' to MCU
[20:25:24][V][miot:091]: Received MCU message 'properties_changed 4 9 0'
[20:25:24][V][miot.binary_sensor:011]: MCU reported sensor 4:9 is: OFF
```
Physically pressing and holding down the feed button on the device.
```
[19:56:38][V][miot:174]: Sending reply 'down none' to MCU
[19:56:38][V][miot:091]: Received MCU message 'event_occured 4 3 5 255'
[19:56:38][W][miot:297]: Unknown command 'event_occured'
[19:56:38][V][miot:174]: Sending reply 'ok' to MCU
[19:56:38][V][miot:091]: Received MCU message 'get_down'
...
[19:56:45][V][miot:174]: Sending reply 'down none' to MCU
[19:56:45][V][miot:091]: Received MCU message 'event_occured 4 2 4 3 5 255'
[19:56:45][W][miot:297]: Unknown command 'event_occured'
[19:56:45][V][miot:174]: Sending reply 'ok' to MCU
[19:56:45][V][miot:091]: Received MCU message 'get_down'
[19:56:45][V][miot:174]: Sending reply 'down none' to MCU
[19:56:45][V][miot:091]: Received MCU message 'event_occured 4 1 4 3 5 255'
[19:56:45][W][miot:297]: Unknown command 'event_occured'
[19:56:45][V][miot:174]: Sending reply 'ok' to MCU
[19:56:45][V][miot:091]: Received MCU message 'properties_changed 4 9 0'
[19:56:45][V][miot.binary_sensor:011]: MCU reported sensor 4:9 is: OFF
[19:56:45][V][miot:174]: Sending reply 'ok' to MCU
```
Pressing for even longer
```
[20:10:07][V][miot:091]: Received MCU message 'event_occured 4 3 5 255'
[20:10:07][W][miot:297]: Unknown command 'event_occured'
[20:10:07][V][miot:174]: Sending reply 'ok' to MCU
[20:10:07][V][miot:091]: Received MCU message 'get_down'
...
[20:10:16][V][miot:174]: Sending reply 'down none' to MCU
[20:10:16][V][miot:091]: Received MCU message 'event_occured 4 2 4 5 5 255'
[20:10:16][W][miot:297]: Unknown command 'event_occured'
[20:10:16][V][miot:174]: Sending reply 'ok' to MCU
[20:10:16][V][miot:091]: Received MCU message 'get_down'
[20:10:16][V][miot:174]: Sending reply 'down none' to MCU
[20:10:16][V][miot:091]: Received MCU message 'event_occured 4 1 4 5 5 255'
[20:10:16][W][miot:297]: Unknown command 'event_occured'
[20:10:16][V][miot:174]: Sending reply 'ok' to MCU
[20:10:16][V][miot:091]: Received MCU message 'properties_changed 4 9 0'
[20:10:16][V][miot.binary_sensor:011]: MCU reported sensor 4:9 is: OFF
```

I'll be available for testing this on the real device if/when this is implemented.

dhewg, 2024-02-01:
It's not part of the TODO, I've seen that command as documented, but my device just doesn't make use of it ;)

You can see the defined events in the json (the link at the top) (as well as the SIID 1):
https://miot-spec.org/miot-spec-v2/instance?type=urn:miot-spec-v2:device:pet-feeder:0000A06C:mmgg-fi1:1

So `event_occured 4 3 5 255` is `feedstart` with `outfood-id=255`.
`event_occured 4 2 4 1 5 255` would be `feedstats` with `outfood-num=1` and `outfood-id=255`.
And `event_occured 4 1 4 1 5 255` finally `feedsuccess` with the same arguments.

The better question is: what do we do with these events?

dhewg, 2024-02-05:
That looks good, thanks! I've not used remote events yet, but I'll take a closer look soon.
Out of curiosity: Do you plan to use that on the HA side?

cristianchelu, 2024-03-16:
> Out of curiosity: Do you plan to use that on the HA side?

> commented on Feb 5

:facepalm: I missed this somehow!

I believe I can use this on HA itself also; for the Mi Home app, my friends used to signal they arrived pet-sitting by pressing the feed button and I'd get a notification on my phone. The same would be doable and useful in HA.

However, handling events themselves via ESPHome automations can also be useful for e.g. tracking "total portions dispensed today", a feature available in the original app but for which an MCU property doesn't exist.

I could add a global integer and increment that with each `feed-finished` event via [globals.set](https://esphome.io/guides/automations.html#globals-set-action), and expose the global as a sensor.

I can't depend on #16 for the above, because that would double-count the portions, the `4:4 feed-portions` property is updated at least twice per feed -- in multiple status and one finished messages.

P.S.: I have a feeling this device uses every little last feature of the miot protocol! :smile:

dhewg, 2024-09-25:
Resolved with #16

closed 2024-09-25 by dhewg