dhewg/esphome-miot #6: Incorrect parsing of `result` from `action` command

cristianchelu, 2024-02-04:
When we recieve a `result` message after dispatching an `action`, it's in a different form than `get_properties`/`set_properties`/`properties_changed`.
More specifically, there is only one `SIID`, one `AIID` and one `code`, but then multiple `piid`+`value` pairs belonging to the same service, and the `update_properties` function expects the pairs to always be `siid`+`piid`

I believe the same will happen for `event_changed` messages when those are implemented.

For convenience:
[Get Properties Spec](https://github.com/blakadder/miot?tab=readme-ov-file#get_properties)
```
    format：get_properties <siid> <piid> ... <siid> <piid>
    Return value：result <siid> <piid> <code> [value] ... <siid> <piid> <code> [value]
```
[Set Properties Spec](https://github.com/blakadder/miot?tab=readme-ov-file#get_properties)
```
    format： set_properties <siid> <piid> <value> ... <siid> <piid> <value>
    Return value： result <siid> <piid> <code> ... <siid> <piid> <code>
```
[Action Spec](https://github.com/blakadder/miot?tab=readme-ov-file#get_properties)
```
    format：action <siid> <aiid> <piid> <value> ... <piid> <value>
    Return value：result <siid> <aiid> <code> <piid> <value> ... <piid> <value>
```

---
I pushed a fix in my repo [here](https://github.com/dhewg/esphome-miot/compare/main...cristianchelu:esphome-miot:fix/parse-action-results), but I'm a noob at C++ and figured you might have a better approach to this.

main branch:
```
[20:20:48][D][miot:125]: Queuing MCU command 'action 5 6'
[20:20:49][V][miot:091]: Received MCU message 'get_down'
[20:20:49][V][miot:171]: Sending reply 'down action 5 6' to MCU
[20:20:49][V][miot:091]: Received MCU message 'result 5 6 0 12 "0,255,255,255,255,1,255,255,255,255,2,8,0,5,0,3,255,255,255,255,4,255,255,255,255"'
[20:20:49][W][miot:183]: Received property value without component: 5 6 12
[20:20:49][V][miot:171]: Sending reply 'ok' to MCU
```
My branch:
```
[20:13:21][D][miot:125]: Queuing MCU command 'action 5 6'
[20:13:22][V][miot:091]: Received MCU message 'get_down'
[20:13:22][V][miot:171]: Sending reply 'down action 5 6' to MCU
[20:13:22][V][miot:091]: Received MCU message 'result 5 6 0 12 "0,255,255,255,255,1,255,255,255,255,2,8,0,5,0,3,255,255,255,255,4,255,255,255,255"'
[20:13:22][V][miot.text_sensor:011]: MCU reported text sensor 5:12 is: 0,255,255,255,255,1,255,255,255,255,2,8,0,5,0,3,255,255,255,255,4,255,255,255,255
[20:13:22][V][text_sensor:013]: 'Feed Plan': Received new state 0,255,255,255,255,1,255,255,255,255,2,8,0,5,0,3,255,255,255,255,4,255,255,255,255
[20:13:22][D][text_sensor:064]: 'Feed Plan': Sending state '0,255,255,255,255,1,255,255,255,255,2,8,0,5,0,3,255,255,255,255,4,255,255,255,255'
[20:13:22][V][miot:171]: Sending reply 'ok' to MCU
```
YAML (both):
```yaml
  - platform: "miot"
    miot_siid: 5
    miot_piid: 12
    miot_poll: false
    name: "Feed Plan"
    icon: mdi:calendar
 ```

dhewg, 2024-02-04:
Nice find, it's unfortunate that both commands return `result` instead of the `action` one e.g. `action_result`.
Since the actions of my device just return a code, does this patch fix it? 
[0001-properly-parse-action-results.patch.txt](https://github.com/dhewg/esphome-miot/files/14158077/0001-properly-parse-action-results.patch.txt)

dhewg, 2024-02-05:
Nice, thanks for the test drive. Fix pushed to main