dhewg/esphome-miot #14: Xiaomi Air Purifier 4 Lite (rma2) support

gvillo, 2024-03-12:
Hi! I was wondering if it supports `zhimi.airp.rma2`, I don't know exactly what are the differences with the supported `rmb1`

kyutov, 2024-03-14:
There are small difference between the two models. You can compare the JSONs with the specs of the two models and see the differences.

[zhimi-rmb1 specs](https://miot-spec.org/miot-spec-v2/instance?type=urn:miot-spec-v2:device:air-purifier:0000A007:zhimi-rmb1:2)
[zhimi-rma2 specs](https://miot-spec.org/miot-spec-v2/instance?type=urn:miot-spec-v2:device:air-purifier:0000A007:zhimi-rma2:1)

helgek, 2024-03-14:
@kyutov Do you maybe know where I can see which revision my device is after flashing ESPHome? I flashed the device straight without checking beforehand. Just asking in case it is the zhimi.airp.rma2 revision I could then confirm that it works with the zhimi.airp.rmb1.yaml config.

dhewg, 2024-03-15:
Yeah, there're some property differences, so that one needs its own config.
If you have a backup of the vendor fw, you should find the model as string in there

helgek, 2024-03-16:
@dhewg Thank you, yes I was able to find the ot_config.psm_modelzhimi.airp.rmb1 string in the backup of the original firmware. So I was lucky to have the right device for the already existing ESPHome configuration.

dhewg, 2025-02-08:
Closing, this never was about `rma2`, but the already supported `rmb1`.

closed 2025-02-08 by dhewg