dhewg/esphome-miot #18: MCU time request: no time source available

kyutov, 2024-04-12:
I have flashed successfully Xiaomi Air Purifier 3H (zhimi.airpurifier.mb3) and everything works without any issues. 

While I was monitoring the logs, I saw that the below error occurs very often: 

`[17:37:38][W][miot:295]: MCU time request: no time source available
`

It doesn't cause any negative effect, but I wanted to make sure that there are no issues before I prepare a PR with the config file and the wiki page for that particular model.

Does anyone know what can cause those errors?

cristianchelu, 2024-04-13:
That's completely harmless.

The microcontroller wants to know the date & time, but the esphome config doesn't contain that info by default.

If there's no clock display on the device or schedules that are configured on-device to work without internet connectivity (*not* if-this-then-that or app smart scenes) there's no real reason to provide the time in the config.

If you want to slience the warning for the sake of silencing it, or you have a different device that needs to know the time, add a [Time Component](https://esphome.io/components/time/index.html) block to your yaml:
```yaml
# ...
time:
  - platform: homeassistant
      id: homeassistant_time
      timezone: GMT0
# ...
```

If your MCU time request message is `time posix` then the `timezone:` variable doesn't matter, it will always receive the UTC / GMT+0 time. If it's just `time` then you can configure the `timezone:` in your local TZ, per [the specs](https://esphome.io/components/time/index.html#configuration-variables).

Enjoy your liberated device!

helgek, 2024-04-13:
@cristianchelu the Mi Smart Fans I just started to build a config for do have a schedule function. I certainly won't use it for myself in a Home Assistant environment but for the sake of completeness of functionality this would probably be required.

cristianchelu, 2024-04-13:
@helgek looking at both your devices' miot spec pages I see nothing that would suggest an on-device schedule. 

The closest property is SIID2:PIID6 [off-delay-time](https://miot-spec.org/miot-spec-v2/spec/property?type=urn:miot-spec-v2:property:off-delay-time:00000054) for the regular and SIID3:PIID1 [off-delay-time](https://miot-spec.org/miot-spec-v2/spec/property?type=urn:dmaker-spec:property:off-delay-time:00000001) for the Pro.

I assume that is specified as a countdown of 0 ~ 480 minutes from when you set it.

If there's a "Run at 2pm for 1h every day" function in the app it's most likely a xiaomi server somewhere sending a turn on command at that hour and a turn off command an hour later, etc -- which means this schedule won't work if your internet connection is down at those specific moments.

See my [mmgg.feeder.fi pull request](https://github.com/dhewg/esphome-miot/pull/17) for an example of a device that needs to know the time. The relevant siid is `Feeding Schedule (SIID: 5)`. 

In my case, the feeder is configured dispense at 8am and 12pm even with the internet down (and via batteries also with the power out :slightly_smiling_face:) via the 5 - feedlistadd, 6 - feedlistedit, 7 - feedlistdel actions.

helgek, 2024-04-13:
You're right, I'm sorry Cristian. Shouldn't write posts immediately after waking up after not having slept much. Indeed there's no such property. I was probably triggered by the ESPHome log of the device where I think the MCU also asks for the time, but probably also for no necessary reason :-D

cristianchelu, 2024-04-13:
Haha, I know that feeling; absolutely no worries!

Apparently every device encountered so far requests the time by default, whether it has a use for it or not.

In any case, @kyutov, if you're satistifed with the above explanations you can close this issue and submit your PR with or without the time block.

kyutov, 2024-04-16:
@cristianchelu - thank you for your assistance! I have added the suggested time component in the YAML file and now the error is silenced! I have prepared the following PR for my changes: #20 and I'm closing the issue.

closed 2024-04-16 by kyutov