dhewg/esphome-miot #21: After a day or so "Network status changed to 'cloud'" - unable to control the unit

smogger, 2024-04-30:
Got 2 Air Purifier 4 Lite, Firmware updated both of them, but one of the unit keeps "disconnecting" from the network (like the Wifi signal goes off on the OLED display)

Still, I am able to update the firmware via Wifi-OTA from ESPHome, it is "alive", but no information sent to ESPHome. Based on log the device goes into "cloud mode" which as I remember happened similarly when I blocked the net with the original firmware. (now there is no block on these models).

When I do a power cycle it start sending again data, can control it and so on, but after a while It goes off again. First I set up a reboot upon network lost, but obviously this is not the case here since I can still access it and can see the log files. 

Attached the latest log. (bedroom which do this issue) and for reference the good one (living room)
[logs_air-purifier-bedroom_run.txt](https://github.com/dhewg/esphome-miot/files/15168626/logs_air-purifier-bedroom_run.txt)
[logs_air-livingroom_logs.txt](https://github.com/dhewg/esphome-miot/files/15168646/logs_air-livingroom_logs.txt)

Thanks

dhewg, 2024-09-25:
The logs don't have the relevant parts in there. You can try setting the logger to VERBOSE and then you'll see when HA is connecting to it.
Generally it sounds like the router disconnects the device from the network, you may have to set some options for compatibility on the wifi net to keep it stable. For that reason I've a dedicated "iot" wifi net

dhewg, 2025-02-08:
Could you resolve the issue? Maybe building a firmware with the latest ESPHome helps?

I have the same device, working without any hiccups since the creation of these components ;)

dhewg, 2025-02-14:
Closing this, I don't believe there's an issue on our end, but keep us updated!

closed 2025-02-14 by dhewg