dhewg/esphome-miot #22: Suport for Smart Purifier 4 Compact (board AC-M18-SC)

tujdfhxyvweyka5, 2024-05-18:
Hello, 

I have a Air Purifier 4 Compact (zhimi.airp.cpa4) with a ESP32-WROOM-32D that I successfully flashed using the toolchain. However, it doesn't seem to work as expected. At first, I noticed that the buttons didn't light and the initial beep would not be heard as with the stock firmware. 

In the logs, I noticed that the firmware would constantly repeat over and over the same sequences:

```
[21:48:16][D][miot:139]: Queuing MCU command 'get_properties 2 1 2 2 2 4 3 4 4 1 4 3 4 4 6 1 8 1 9 1 9 11 13 2'
[21:48:39][D][miot:139]: Queuing MCU command 'set_properties 11 4 60'
[21:49:16][D][miot:139]: Queuing MCU command 'get_properties 2 1 2 2 2 4 3 4 4 1 4 3 4 4 6 1 8 1 9 1 9 11 13 2'
[21:49:39][D][miot:139]: Queuing MCU command 'set_properties 11 4 60'
```

It seems no action are being taken into account. 

I was wondering if the communication with the MCU is operational. I noticed that if I invert the UART pins for MCU serial communication I obtain the same output. 

For those who are interested, here's a picture of my board, with the mandatory flashing pins highlighted. 

[image]

I also join this this issue the stock firmware logs and the complete ESPHome firmware logs:

[logs_zhimi.airp.cpa4_run.log](https://github.com/dhewg/esphome-miot/files/15360371/logs_zhimi.airp.cpa4_run.log)
[stock_firmware_ac-m18-sc.log](https://github.com/dhewg/esphome-miot/files/15360372/stock_firmware_ac-m18-sc.log)

Also the MIOT specs:

[https://home.miot-spec.com/spec/zhimi.airp.cpa4](https://home.miot-spec.com/spec/zhimi.airp.cpa4)

Jaco1990, 2024-05-29:
The Xiaomi Compact 4 Air Purifier does not have a secondary MCU onboard. Instead every thing is controlled by the onboard ESP32. There is no individual integrations in ESPhome for the components of this humidifier.

tujdfhxyvweyka5, 2024-10-09:
OK thanks for your feedback @Jaco1990 so basically it means that reverse engineering it is the only way ?

dhewg, 2025-02-08:
That's unfortunate :(

It means that the components here are of no use.

But at least it's an ESP32 and not some unsupported hardware by PlatformIO/ESPHome. So a ESPHome firmware is still doable, but how useful that is depends on the actually used sensors etc. and if those are supported by ESPHome

closed 2025-02-08 by dhewg

simonepsp, 2025-12-10:
> Hi, I just created an [ESPHome package](https://github.com/m5k10/esphome-xiaomi-compact-4) that allows you to replace the original firmware. Not everything works, but I think it’s worth trying. At the very least, the code can serve as pinout documentation for anyone who wants to write their own version.

Thank you for working on this device. I'm interested in getting one as well. 
What's the status of the integration? What's still not working?

foxalite, 2026-10-03:
> I don't know what board revision is pictured above, but revision 1.3 contains both an ESP32 and Gigachip MCU. I have been able to successfully flash ESPhome, and everything works except for:
> 
>     1. Fan speed - reads as `Unknown` in HA
> 
>     2. ~Fan Control - Works to set, but state tracking is messed up.~
> 
.....
> 
> Update: Fan control reporting does work, just needed the option to be blank.

Hello,
Here is my YAML.
The fan speed can be changed in manual mode (from 1 to 14).

```
external_components:
  - source:
      type: git
      url: https://github.com/dhewg/esphome-miot
      ref: main
    components: [ miot ]

uart:
  id: miot_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200

miot:
  id: miot_main
  uart_id: miot_uart
  miot_heartbeat_siid: 11
  miot_heartbeat_piid: 4
  miot_ota_net_indicator: ok

fan:
  - platform: "miot"
    name: "Fan"
    state:
      miot_siid: 2
      miot_piid: 1
    speed:
      miot_siid: 2      # Parfois SIID 9 selon la variante, commencez par le SIID 2
      miot_piid: 1     # Propriété "favorite-level"
      min_value: 1      # Obligatoire : Valeur minimale du niveau favori
      max_value: 1     # Obligatoire : Valeur maximale du niveau favori
    preset_modes:
      miot_siid: 2
      miot_piid: 4
      options:
        0: "Auto"
        1: "Sleep"
        2: "Favorite"
        3: ""
number:
  - platform: "miot"
    name: "Vitesse Mode Favorite"
    miot_siid: 9        # L'ID de service custom du constructeur zhimi
    miot_piid: 11       # La propriété "favorite-level" du Compact 4
    min_value: 1
    max_value: 14       # Accepte les valeurs de 1 à 14
    step: 1
    
sensor:
  - platform: "miot"
    name: "Qualité de l'air PM2.5"
    icon: "mdi:blur"
    miot_siid: 3
    miot_piid: 4
    unit_of_measurement: "µg/m³"
```

Jchisholm204, 2026-11-03:
@foxalite In your configuration, do you see any MCU reset errors? Also, do you observe the used time and filter status change over time? 

When I tried to implement #113, I ran into an issue where, with the configuration you're using now, the Gigachip MCU could only work with so many MIOT entities at a given time, resulting in some soft rebooting issues.

foxalite, 2026-11-11:
> [@foxalite](https://github.com/foxalite) In your configuration, do you see any MCU reset errors? Also, do you observe the used time and filter status change over time?
>
> When I tried to implement [...]

this is my full yaml:

```
esphome:
  name: purificateur-compact-4
  friendly_name: Xiaomi Compact 4
 
  includes:
    - mon_code.h

esp32:
  board: esp32dev
  variant: ESP32
  flash_size: 4MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: "y"
      CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE: "n"
      # --- CORRECTIFS VITAUX POUR L'AUTHENTIFICATION WI-FI XIAOMI ---
      CONFIG_ESP32_CHECK_EFUSE_MAC_CRC: "n"
      CONFIG_SOC_NOT_CHECK_EFUSE_MAC_CRC: "y"
    advanced:
      ignore_efuse_mac_crc: true
      ignore_efuse_custom_mac: true

# Votre Wi-Fi privé (sans accès internet requis)
wifi:
  id: wifiClient
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: Xiaomi 4 lite Fallback Hotspot
    password: "qRqu8sG7BkAK"

# L'API locale pour Home Assistant
api:
  encryption:
    key: !secret purificateur_compact_4__api

logger:
  level: DEBUG

ota:
  - platform: esphome

web_server:
  port: 80
  include_internal: True

safe_mode:
  disabled: true
  
external_components:
  - source:
      type: git
      url: https://github.com/dhewg/esphome-miot
      ref: main
    components: [ miot ]

uart:
  id: miot_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200

miot:
  id: miot_main
  uart_id: miot_uart
  miot_heartbeat_siid: 11
  miot_heartbeat_piid: 4
  miot_ota_net_indicator: ok

fan:
  - platform: "miot"
    name: "Fan"
    state:
      miot_siid: 2
      miot_piid: 1
    speed:
      miot_siid: 2      # Parfois SIID 9 selon la variante, commencez par le SIID 2
      miot_piid: 1     # Propriété "favorite-level"
      min_value: 1      # Obligatoire : Valeur minimale du niveau favori
      max_value: 1     # Obligatoire : Valeur maximale du niveau favori
    preset_modes:
      miot_siid: 2
      miot_piid: 4
      options:
        0: "Auto"
        1: "Sleep"
        2: "Favorite"
        3: ""
number:
  - platform: "miot"
    name: "Vitesse Mode Favorite"
    miot_siid: 9        # L'ID de service custom du constructeur zhimi
    miot_piid: 11       # La propriété "favorite-level" du Compact 4
    min_value: 1
    max_value: 14       # Accepte les valeurs de 1 à 14
    step: 1
    
sensor:
  - platform: "miot"
    name: "Qualité de l'air PM2.5"
    icon: "mdi:blur"
    miot_siid: 3
    miot_piid: 4
    unit_of_measurement: "µg/m³"

```

the include (mon_code.h) .
```
#pragma once
#include <deque>
```

Unfortunately, I haven't had time to go any further. I don't received any feedback on the filter's status.
If I try your configuration, the purifier disconnects from the Home Assistant API every 2 seconds, rendering it unusable.

Another point: the photo shows the older version, without the 2 MCUs.
This is mine:
To enter in boot, I need to create a bridge between terminal GP00 and ground.
[image]

Other information I received from the log:

```
[14:06:09.261][W][miot:401]: Unknown command 'country_code'
[14:06:15.155][W][miot:259]: Received property value without component: 4 1 72
[14:06:15.256][W][miot:259]: Received property value without component: 4 3 1976
[14:06:15.358][W][miot:259]: Received property value without component: 4 4 210
```

commit 0573c28e (2025-02-08): docs: add zhimi.airp.cpa4 to unsupported devices

Closes #22