dhewg/esphome-miot #23: Compile error

moto25, 2024-06-24:
Hi, since ESPHome 2024.6.0 are OTA changes. When setting

ota:
  - platform: esphome  
    password: !secret ota_password

got compile error:

src/esphome/components/miot/miot.cpp:18:10: fatal error: esphome/components/ota/ota_component.h: No such file or directory
 #include "esphome/components/ota/ota_component.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/mi-gbr/src/esphome/components/miot/miot.o] Error 1

any idea? thank you

scrampker, 2024-06-26:
This is a change to ESPHome. You would update that section to:

ota:
  platform: esphome

That being said, I'm still getting the following afterwards, which means the actual libraries need to be updated:
[image]

9R, 2024-07-08:
#25 is an alternate solution for this issue that makes use of the new esphome function `get_global_ota_callback()`

dhewg, 2024-09-19:
I just merged the fix, thanks everybody!

closed 2024-09-19 by dhewg