dhewg/esphome-miot #26: Notes on Smartmi Evaporative Humidifier 2 (zhimi.humidifier.ca4)

szymucha94, 2024-07-13:
Pins:
[image]
Do not power up the board via serial adapter. Use standard power brick provided with humidifier.
Bridge gpio0 and gpio2, then connect gpio2 to ground (for example gnd for STM serial)
Mobo is sealed with some slimy substance to prevent corrosion due to the vapor, hence little bit more heat for soldering is useful. I didn't find any test point that is connected gpio2. There is tiny gpio0 TP on the other side of mobo but it's not a hole and it's pretty far from gnd, so it's easier to solder all at the same side of board.
Working (for me) esp config. Replace "xxx" with your keys or just use the variable from secret.yaml.
Water level readings are broken for me, but this is due to long dead (probably shorted) sensor above the power port. Typical failure of this model. Speaking from experience if you get values above 100% (or "unknown" state in HA) that either means your sensor is dead or it's gonna die soon.
Source is temporary because repo's version doesn't compile. It's taken from one of the comments.

```
# https://home.miot-spec.com/spec/zhimi.humidifier.ca4

external_components:
  source: github://pchmura4/esphome-miot@main

esphome:
  name: living_room_air_humidifier
  friendly_name: Living Room Air Humidifier
  comment: Smartmi Evaporative Humidifier 2 (zhimi.humidifier.ca4)
  project:
    name: "dhewg.esphome-miot"
    version: "zhimi.humidifier.ca4"

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true

logger:
  level: INFO

api:
  encryption:
    key: "xxx"
  reboot_timeout: 0s
  services:
    - service: mcu_command
      variables:
        command: string
      then:
        - lambda: 'id(miot_main).queue_command(command);'

ota:
 - platform: esphome
   password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Mihumidifier Fallback"
    password: "TuAkop2DnYou"

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200

miot:
  id: miot_main

switch:
  - platform: "miot"
    miot_siid: 2
    miot_piid: 1
    name: "Power"
    icon: mdi:power
  - platform: "miot"
    miot_siid: 2
    miot_piid: 8
    name: "Dry"
    icon: mdi:water-minus
  - platform: "miot"
    miot_siid: 4
    miot_piid: 1
    name: "Notification Sounds"
    icon: mdi:volume-high
    entity_category: config
  - platform: "miot"
    miot_siid: 6
    miot_piid: 1
    name: "Child Lock"
    icon: mdi:lock
    entity_category: config
  - platform: "miot"
    miot_siid: 7
    miot_piid: 5
    name: "Clean Mode"
    icon: mdi:water-minus
    entity_category: config

select:
  - platform: "miot"
    miot_siid: 2
    miot_piid: 5
    name: "Fan level"
    icon: mdi:fan
    options:
      0: "Auto"
      1: "Low"
      2: "Medium"
      3: "High"
  - platform: "miot"
    miot_siid: 5
    miot_piid: 2
    name: "Display brightness"
    icon: mdi:brightness-6
    entity_category: config
    options:
      0: "Dark"
      1: "Glimmer"
      2: "Brightest"

sensor:
  - platform: "miot"
    miot_siid: 2
    miot_piid: 2
    name: "Device Fault Code"
    icon: mdi:fan-alert
    entity_category: diagnostic
  - platform: "miot"
    miot_siid: 3
    miot_piid: 7
    name: "Temperature"
    unit_of_measurement: "°C"
    device_class: temperature
    state_class: "measurement"
    icon: mdi:thermometer
  - platform: "miot"
    miot_siid: 3
    miot_piid: 9
    name: "Relative Humidity"
    unit_of_measurement: "%"
    device_class: humidity
    state_class: "measurement"
    icon: mdi:water-percent
  - platform: "miot"
    miot_siid: 2
    miot_piid: 7
    name: "Water Level"
    unit_of_measurement: "%"
    state_class: "measurement"
    icon: mdi:water-percent
    filters:
      - clamp:
          min_value: 0
          max_value: 100
          ignore_out_of_range: true
  - platform: "miot"
    miot_siid: 2
    miot_piid: 11
    name: "Motor Speed"
    unit_of_measurement: "rpm"
    icon: mdi:fan
  - platform: "miot"
    miot_siid: 7
    miot_piid: 1
    name: "Actual Speed"
    unit_of_measurement: "rpm"
    icon: mdi:fan
  - platform: internal_temperature
    name: "MCU Temperature"
```

dhewg, 2024-09-19:
Nice!
The ota fix is finally merged by now, do you want to open a PR with a config and README entry?

szymucha94, 2024-09-25:
@dhewg PR ready: #32

dhewg, 2024-09-26:
PR is merged, do you feel like adding the instructions/hints/notes to the wiki? ;)

frank8the9tank, 2024-10-11:
I have the same dead level sensor, thus want to flash this, 

1. Solder GPIO0 to GPIO2, check.
2. Solder GPIO2 to GND, check.

Question: 
3. Where do i connect the other 4 pads to on my flasher?

szymucha94, 2024-10-11:
UART adapter TX to ESP RX, UART adapter RX to ESP TX, adapter GND to any of the two GND.

frank8the9tank, 2024-10-11:
So for flashing i don't need to do anything with STM TX and STM RX?
If not, what is the function for these 2 pins?, as they are marked on the picture.

szymucha94, 2024-10-11:
No, you don't need STM TX and RX.
It's for talking to the other controller on the board, ie. the one that controls motors, sensors, display, etc.
ESP is only a cloud bridge and it gets these values from STM32.

frank8the9tank, 2024-10-12:
So i got esphome flashed and connected to HA. When i switch the power ON and OFF i can see the response on the display of the humidifier. But still i cannot get the device running/blowing.
I flashed mine bequase of a 0% water level reading. 
Now with ESPhome on it i still cannot get it running, did you see the same behaviour?
(before, when there was still xiaomi in the esp, the 0% level sensor prevented the fan from blowing, making the device useless)

szymucha94, 2024-10-12:
No, because my "dead" water level sensor was in fact showing always 100% reading, not 0%.
If yours is showing 0% and not "unknown" value that means sensor is not dead, logically speaking at least. Water level sensor is talking to STM32 controller via i2c and if it really died that would result in failure to power on. ESP32 doesn't control this behavior and cannot force STM to turn on the motor.

But...

1. Does your humidifier react to disconnecting water tank?
2. Did you try cleaning two pogo pins that sticks out of the water tank?
3. What about cleaning the water level sensor compartment? The one under plastic shell within water tank. It's recommended to remove all the residue from both pipes.
4. What about the workaround of just bridging both pogo pins? That should give you consistent reading of 100%.

Now, if none of these worked this is how I fixed mine:
I removed the water level sensor from humidifier (it's behind power connector), placed it on a metal plate and I applied about 200C heat via gas soldering iron without a tip to it's black seal residue (not the plastic casing) for about 3 minutes to the point where it started to crack and smoke a little bit.
This effectively fixed broken solder joints inside of the plastic shell without scraping off the seal which would result in tearing off internal components anyway.
It's not a nice solution but it worked for me and the humidifier is reporting valid water level values for months now.
I think that it's easy to kill the sensor if you take off water tank while whole unit is powered on 4-5 times within short period of time.
This is why if you reanimate yours I strongly advise to always disconnect unit from power before you disconnect water tank.

frank8the9tank, 2024-10-13:
I did the 4 point you mentioned already, as a last resort i flashed esphome just to run the motor without the level sensor.
But that did not made a difference.

But

The heating the sensor for 3 minutes did the trick. Good advice. Now have a working level sensor again.
I just a hot air gun with a small nozzle. heated it for 3 minutes and let is cool down for 10 minutes.
It cracked the black still and browned the plastic a bit. But it did the trick.

Many thanks for the tips, 

I also came across this:
https://www.youtube.com/watch?v=Pm4rbp3xFk0
https://github.com/tomasvilda/humidifierstart
But its for a different humidifier type. But maybe the same i2c sensor.

Sirse, 2024-10-16:
Today opened my zhimi.humidifier.cb1 (version without screen) and find out proprietary MHCWB2P :(

I hope someday somebody will create drop-in PCB replacement for those modules.

szymucha94, 2024-10-16:
> Today opened my zhimi.humidifier.cb1 (version without screen) and find out proprietary MHCWB2P :(
> 
> I hope someday somebody will create drop-in PCB replacement for those modules.

Old Marvell controllers aren't that bad. At least they don't disconnect from wifi every 15 minutes when you block their MAC address from accessing WAN.
Btw. if your humidifier has STM32 you can easily replace proprietary MCU with wroom-32d. You just need to (visually?) identify which pins are for the STM's UART.

DmitryZemlyuk, 2024-11-01:
Hi everyone. Can someone send me the original firmware (stock firmware original-firmware.bin)? I need to flash another humidifier to revive it.

dhewg, 2025-02-08:
I added zhimi.humidifier.cb1 to the list of unsupported devices.

It'll be stellar if someone adds a wiki page for zhimi.humidifier.ca4!

closed 2025-02-08 by dhewg