dhewg/esphome-miot #33: Mi Smart Standing Fan 2 Lite - dmaker.fan.1c Support

The-May, 2024-10-08:
Hi I am trying to contribute with my Information/hardware to this awesome project.

Hardware is:  Mi Smart Standing Fan 2 Lite:
https://home.miot-spec.com/spec?type=urn:miot-spec-v2:device:fan:0000A005:dmaker-1c:1
I managed to disassemble the fan and could find a esp on a Mainboard:

The Mainboard revision is MJFS-P12E B1-CTRL-V1.0 and seems to be assembled on 2021.04.06

See PCB Upper side:
[image]

ESP32 is a ESP-WROOM-02D:

[image]

PCB bottom side:

[image]

Please advise and feel free what to do next to flash esphome on it and which config might help, since im not experienced with esphome at all...also i could not find any docs on how to flash the esp32.....thx!

ronnieSVK, 2025-01-01:
Hi @The-May !

I recently flashed the same model. However, if you are not skilled in soldering, microcontrollers and ESPHome im unsure if this is good idea.

helgek, 2025-01-01:
You don’t need to solder. I flashed more than a dozen devices using PCBite from Sensepeek. Another cheap solution is using e.g. acupuncture needles with e.g. plasticine as holder. There are also 3d printed examples of holders.

regarding flashing there’s many tutorials available, e.g. check for Syssi‘s yeelight GitHub page but also the Espressif docs for the individual ESP32 versions contain helpful information.

ronnieSVK, 2025-01-01:
Yes, you dont need to solder, if you have prepared some minilab at home with all kind of equipment with 3D printed holders, needles etc... I flashed and developed equally - dozen of devices, and i always prefer soldering. It is the quickest and most stable way to have "contact". There is no need to check for some Yeelight documentations. The question was specifically about "Mi Standing Fan 2 Lite" and all informations are listed on this page. What he needs is ESPHome config file which has some changes in comparison with not "Lite" version.

cristianchelu, 2025-01-01:
@ronnieSVK The wiki is open to edits, so if you by chance snapped any pictures while you flashed it that show what to connect where it should already be helpful to others :)

The same again for the config -- open either a PR with it or just paste it in this thread; others with a bit more knowledge of soldering can make good use of it.

As for @The-May 's original question:

I'm sorry that this device is nowhere near beginner level (hardware wise) to convert :( lemme start by saying:

> [!CAUTION]
> If this is your first time working with electronics, don't do anything with this device without a friend that knows electronics present. Because this fan is connected to mains electricity, if you solder something where you shouldn't, you can start a fire. Or you can end up in the hospital, or worse. And definitely don't do anything to the board while it's plugged in. We want to stay safe while enjoying our hobby.

#### For the chip: 

You can find the [ESP-WROOM-02D](https://www.espressif.com/sites/default/files/documentation/esp-wroom-02u_esp-wroom-02d_datasheet_en.pdf)'s datasheet by searching online, scroll to page 3 for the pinout.

#### For flashing

Also by searching "esphome flashing", you can find [this guide](https://esphome.io/guides/physical_device_connection.html#module-only) or plenty of different tutorials, but flashing a new board still requires some patience and ingenuity.

At the very least you will require a multimeter and a soldering iron, besides the USB<->UART flasher.

I've overlaid the pinout from the schematic over your picture and attempted to figure out what pins are broken out where:
(Most of the time, but not always, the designers add through-hole connectors, or at least pads so they don't have to touch the tiny pins of the chip while they debug their design)

[image]

Practically, you only need:
- `GND`, `3.3V (3V3)` for power
-  `RX` and `TX` for flashing (pay attention to cross-connect, RX->TX, TX->RX)
-  and `IO0` connected to a `GND` when you first apply 3.3V to put the chip in programming mode, which you can then remove after a few seconds.

Reallistically, you will need to use a multimeter on continuity mode to figure out where, **if at all** all the pins are connected to a test pad.

I'm only confident in `GND` and `TX` (by the label `UART0_TXD`).

I **think** I read `RXD` to the right of `UART0_TXD` so that might be a candidate to check with a multimeter if it's connected to the right pin on the chip.

`3V3` and `IO0` I don't know, good luck checking, they might be somewhere.

If you can't find a beep anywhere -- I'm sorry, the only way would be to solder, use pcbite/acupuncture needles/tools/magic to connect straight to those tiny pins on the chip - and blame the designers in the process for not making it easier.

#### Actual flashing

1. Install ESPTool.py - [DOCS](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html)
2. Read the flash first and save it somehwere - [DOCS](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#read-flash-contents-read-flash)
3. Generate an ESPHome bare config for this module; besides all the other goodies, it needs to contain:
```yaml
esp8266:
  board: esp_wroom_02
 ``` 
4. Download the `.bin` file from the ESPHome dashboard and flash it with esptool [DOCS](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#write-binary-data-to-flash-write-flash).
4. Check the readme in this project as well as other configs to generate a working one, then post it here or in a PR so people after you won't have to suffer through the entire ordeal from the start :) .

Yeah, you might feel overwhelmed by all this if this is your first project because it's not a beginner task, and that's alright. 

If it's not your first, just first time with ESPHome, sorry for the warnings, but better safe than sorry.

Hope this helps.
Happy new year!

ronnieSVK, 2025-01-01:
Hi @cristianchelu !
Fantastic description !

I have only one-two pictures for my evidence, but you schema is absolutely right !

 - RX + TX are exposed on the board, YELLOW + GREEN on picture. (Honestly, i dont know which is which on the board, since the marks are so small. I flashed also Mi Air Purifier 4 recently, where is the situation similar. So for me it was "hit/miss" while i was watching the USB UART flasher LEDs for feedback.)
 - Since i had no problem with soldering, i wasnt searching if 3.3V is exposed (but i didnt find it with quick look) so i used the ESP PIN marked RED for "it".
 - GND is self-explanatory, on the board
 - GPIO marked GRAY is IO0, which needs to be PULLED DOWN, so again, as you marked, with GND
 
[image]

ESPHome code i wrote (of course, feel free to edit or change it to your needs)

`external_components:
  source: github://dhewg/esphome-miot@main

esphome:
  name: "add_your_device_name"
  friendly_name: "add_your_friendly_name"

esp8266:
  board: esp_wroom_02

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Bratislava
    servers:
     - 0.pool.ntp.org
     - 1.pool.ntp.org
     - 2.pool.ntp.org

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "gX8HHg5BgFWQqKpK6TCzACl6aUSUhZ9lPXrsFvlL3iY="
  reboot_timeout: 0s
  services:
    - service: mcu_command
      variables:
        command: string
      then:
        - lambda: 'id(miot_main).queue_command(command);'

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: True

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom-Mi-Air-Fan-2-Lite"
    password:

captive_portal:

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 115200

miot:
  id: miot_main

switch:
  - platform: "miot"
    id: "on"
    miot_siid: 2
    miot_piid: 1
    name: "Power"
    icon: "mdi:power"
    on_turn_off:
      - lambda: |- 
          if(id(template_fan).state == true) {
            auto call = id(template_fan).turn_off();
            call.perform();
          }
    on_turn_on:
      - lambda: |- 
          if(id(template_fan).state == false) {
            auto call = id(template_fan).turn_on();
            call.perform();
          }
  - platform: "miot"
    id: "horizontal_swing"
    miot_siid: 2
    miot_piid: 3
    name: "Oscillation"
    icon: "mdi:arrow-oscillating"
    on_turn_off:
      - lambda: |-
          if (id(template_fan).oscillating == true) {
            auto call = id(template_fan).make_call();
            call.set_oscillating(false);
            call.set_preset_mode(id(mode).state.c_str());
            call.perform();
          }
    on_turn_on:
      - lambda: |-
          if (id(template_fan).oscillating == false) {
            auto call = id(template_fan).make_call();
            call.set_oscillating(true);
            call.set_preset_mode(id(mode).state.c_str());
            call.perform();
          }

  - platform: "miot"
    miot_siid: 2
    miot_piid: 12
    name: "Indicator Lights"
    icon: "mdi:lightbulb"
    entity_category: config
  - platform: "miot"
    miot_siid: 2
    miot_piid: 11
    name: "Notification Sounds"
    icon: "mdi:volume-high"
    entity_category: config
  - platform: "miot"
    miot_siid: 3
    miot_piid: 1
    name: "Child Lock"
    icon: "mdi:lock"
    entity_category: config
  
select:
  - platform: "miot"
    id: "mode"
    miot_siid: 2
    miot_piid: 7
    name: "Mode"
    icon: "mdi:leaf"
    options:
      0: "Direct Breeze"
      1: "Natural Breeze"
    on_value:
      - lambda: |-
          if(id(template_fan).preset_mode != x) {
            auto call = id(template_fan).make_call();
            call.set_preset_mode(x.c_str());
            call.perform();
          }

number:
  - platform: "miot"
    miot_siid: 2
    miot_piid: 2
    name: "Fan Level"
    icon: "mdi:fan-chevron-up"
    min_value: 1
    max_value: 3
    step: 1
  - platform: "miot"
    miot_siid: 2
    miot_piid: 10
    name: "Off Delay"
    icon: "mdi:clock-outline"
    unit_of_measurement: "min"
    device_class: duration
    min_value: 0
    max_value: 480
    step: 1

button:
  - platform: "miot"
    miot_siid: 2
    miot_aiid: 1
    name: "Toggle Power"
    icon: "mdi:power-cycle"
  - platform: "miot"
    name: "Toggle Mode"
    icon: "mdi:leaf"
    miot_siid: 4
    miot_aiid: 1
  - platform: "miot"
    name: "Toggle Speed"
    icon: "mdi:fan"
    miot_siid: 4
    miot_aiid: 2

fan:
  - platform: "template"
    id: template_fan
    name: None
    has_oscillating: true
    speed_count: 100
    preset_modes:
      - "Direct Breeze"
      - "Natural Breeze"
    restore_mode: NO_RESTORE
    on_turn_on:
      - lambda: if (id(on).state == false) id(on).turn_on();
    on_turn_off:
      - lambda: if (id(on).state == true) id(on).turn_off();
    on_oscillating_set:
      - lambda: |-
          if (id(horizontal_swing).state != x) {
            if (x == true) {
              id(horizontal_swing).turn_on();
            } else {
              id(horizontal_swing).turn_off();
            }
          }
          auto call = id(template_fan).make_call();
          call.set_preset_mode(id(mode).state.c_str());
          call.perform();`

The-May, 2025-01-30:
Update: dumped the og .bin with esptool.
after that I tried flashing it, but it somehow didnt work out and it does not seem to access the network at all.
Debugging seems to be hard too, becasue i cant view the logs.

```
external_components:
  source: github://dhewg/esphome-miot@main

esphome:
  name: esp32-02d-mifan01-c45fab
  friendly_name: esp32-02d-mifan01-c45fab

esp8266:
  board: esp_wroom_02

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Berlin
    servers:
     - 0.pool.ntp.org
     - 1.pool.ntp.org
     - 2.pool.ntp.org

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "changemechangemechangemechangemechangeme"
  reboot_timeout: 0s
  services:
    - service: mcu_command
      variables:
        command: string
      then:
        - lambda: 'id(miot_main).queue_command(command);'

ota:
  - platform: esphome
    password: !secret ota_pw

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: True

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-02D-Mifan01-C45Fab"
    password: !secret fallback_pass

captive_portal:

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 115200

miot:
  id: miot_main

switch:
  - platform: "miot"
    id: "on"
    miot_siid: 2
    miot_piid: 1
    name: "Power"
    icon: "mdi:power"
    on_turn_off:
      - lambda: |- 
          if(id(template_fan).state == true) {
            auto call = id(template_fan).turn_off();
            call.perform();
          }
    on_turn_on:
      - lambda: |- 
          if(id(template_fan).state == false) {
            auto call = id(template_fan).turn_on();
            call.perform();
          }
  - platform: "miot"
    id: "horizontal_swing"
    miot_siid: 2
    miot_piid: 3
    name: "Oscillation"
    icon: "mdi:arrow-oscillating"
    on_turn_off:
      - lambda: |-
          if (id(template_fan).oscillating == true) {
            auto call = id(template_fan).make_call();
            call.set_oscillating(false);
            call.set_preset_mode(id(mode).state.c_str());
            call.perform();
          }
    on_turn_on:
      - lambda: |-
          if (id(template_fan).oscillating == false) {
            auto call = id(template_fan).make_call();
            call.set_oscillating(true);
            call.set_preset_mode(id(mode).state.c_str());
            call.perform();
          }

  - platform: "miot"
    miot_siid: 2
    miot_piid: 12
    name: "Indicator Lights"
    icon: "mdi:lightbulb"
    entity_category: config
  - platform: "miot"
    miot_siid: 2
    miot_piid: 11
    name: "Notification Sounds"
    icon: "mdi:volume-high"
    entity_category: config
  - platform: "miot"
    miot_siid: 3
    miot_piid: 1
    name: "Child Lock"
    icon: "mdi:lock"
    entity_category: config
  
select:
  - platform: "miot"
    id: "mode"
    miot_siid: 2
    miot_piid: 7
    name: "Mode"
    icon: "mdi:leaf"
    options:
      0: "Direct Breeze"
      1: "Natural Breeze"
    on_value:
      - lambda: |-
          if(id(template_fan).preset_mode != x) {
            auto call = id(template_fan).make_call();
            call.set_preset_mode(x.c_str());
            call.perform();
          }

number:
  - platform: "miot"
    miot_siid: 2
    miot_piid: 2
    name: "Fan Level"
    icon: "mdi:fan-chevron-up"
    min_value: 1
    max_value: 3
    step: 1
  - platform: "miot"
    miot_siid: 2
    miot_piid: 10
    name: "Off Delay"
    icon: "mdi:clock-outline"
    unit_of_measurement: "min"
    device_class: duration
    min_value: 0
    max_value: 480
    step: 1

button:
  - platform: "miot"
    miot_siid: 2
    miot_aiid: 1
    name: "Toggle Power"
    icon: "mdi:power-cycle"
  - platform: "miot"
    name: "Toggle Mode"
    icon: "mdi:leaf"
    miot_siid: 4
    miot_aiid: 1
  - platform: "miot"
    name: "Toggle Speed"
    icon: "mdi:fan"
    miot_siid: 4
    miot_aiid: 2

fan:
  - platform: "template"
    id: template_fan
    name: None
    has_oscillating: true
    speed_count: 100
    preset_modes:
      - "Direct Breeze"
      - "Natural Breeze"
    restore_mode: NO_RESTORE
    on_turn_on:
      - lambda: if (id(on).state == false) id(on).turn_on();
    on_turn_off:
      - lambda: if (id(on).state == true) id(on).turn_off();
    on_oscillating_set:
      - lambda: |-
          if (id(horizontal_swing).state != x) {
            if (x == true) {
              id(horizontal_swing).turn_on();
            } else {
              id(horizontal_swing).turn_off();
            }
          }
          auto call = id(template_fan).make_call();
          call.set_preset_mode(id(mode).state.c_str());
          call.perform();
```

ronnieSVK, 2025-01-30:
Hi @The-May !

I start from the point, that your WiFi credentials are defined and stored in !secret.yaml, as the code show. Try to remove `fast_connect` option, and try to add static IP (it causes sometimes issues) regarding to your LAN setup 👍 # Optional 
  
```
manual_ip:
    static_ip: 192.168.0.123
    gateway: 192.168.0.1
    subnet: 255.255.255.0
```

https://esphome.io/components/wifi.html

cristianchelu, 2025-01-30:
Wow, that BOOT0 breakout seems a long way away from the chip, but if it works, it works...

The built-in esphome serial logger is available on the `WIFITXD` breakout test pad, right next to where you put your 3V3 (from the wiki pictures); I see in your YAML that you left the logger enabled:
```yaml
# Enable logging
logger:
# baud_rate: 0 # <-- this line would mean no serial output, without it you have serial log out
```
try to solder an RX line on your UART adapter there and see what comes out.

Also, I see you've copied the fast connect setting from ronnieSVK's config, can you try without it?
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none # <-- delete this
  fast_connect: True # <-- delete this
```

Also, make sure you have your correct secrets for wifi, and if all else fails, leave it for a few minutes and see if the backup AP shows up on your phone, that's another indication the chip works fine just can't connect to your wifi.
```yaml
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-02D-Mifan01-C45Fab"
    password: !secret fallback_pass
```

Is your WiFi visible on scan or hidden, btw? If it's hidden, it won't find it.

Lemme know if this helps.

cristianchelu, 2025-01-30:
Also, both to you and @ronnieSVK , I suggest you disable your built-in serial logger on ESP8266 chips after successful setup by adding:
```yaml
logger:
  baud_rate: 0
```
if you find that it struggles to process miot messages often (`unknown mcu message: get_do??` or some such).

I've found a *ton* of missed serial messages from the MCU with that enabled.

ronnieSVK, 2025-01-30:
Hi @cristianchelu !

Thank you for your suggestion. Yes, it is present also on my side. As usually, i dont need logger, till no issues are present :D I think i will disable the logger completely. Honestly im using only ESP32 chips for my projects and since i had no issues there, i have logger as a standard part... As you raised your finger, this is a good point on 8266, thanks for that !

The-May, 2025-01-30:
> Wow, that BOOT0 breakout seems a long way away from the chip, but if it works, it works...
> 
> The built-in esphome serial logger is available on the `WIFITXD` breakout test pad, right next to where you put your 3V3 (from the wiki pictures); I see in your YAML that you left the logger enabled:
> 
> # Enable logging
> logger:
> # baud_rate: 0 # <-- this line would mean no serial output, without it you have serial log out
> 
> try to solder an RX line on your UART adapter there and see what comes out.
> 
> Also, I see you've copied the fast connect setting from ronnieSVK's config, can you try without it?
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   power_save_mode: none # <-- delete this
>   fast_connect: True # <-- delete this
> 
> Also, make sure you have your correct secrets for wifi, and if all else fails, leave it for a few minutes and see if the backup AP shows up on your phone, that's another indication the chip works fine just can't connect to your wifi.
> 
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "Esp32-02D-Mifan01-C45Fab"
>     password: !secret fallback_pass
> 
> Is your WiFi visible on scan or hidden, btw? If it's hidden, it won't find it.
> 
> Lemme know if this helps.

1.  i followed it, changed to
`# Enable logging
logger:
# baud_rate: 0 # <-- this line would mean no serial output, without it you have serial log out`

2. Deleted lines
  power_save_mode: none # <-- delete this
  fast_connect: True # <-- delete this

3. and yes, secrets work solidly - used them in other projects.

4. regarding wifi:
i have 2 seperate wifis: wifiname (2.4ghz) and wifiname_5 to prevent issues with it (had problems in the past)

The-May, 2025-01-30:
> Hi [@The-May](https://github.com/The-May) !
> 
> I start from the point, that your WiFi credentials are defined and stored in !secret.yaml, as the code shows. Try to remove `fast_connect` option, and try to add static IP (it causes sometimes issues) regarding to your LAN setup :
> 
> ```
> manual_ip:
>     static_ip: 192.168.0.123
>     gateway: 192.168.0.1
>     subnet: 255.255.255.0
> ```
> 
> https://esphome.io/components/wifi.html

have setup a static ip beside a static dhcp-adress.
Still no success.

Just to clarify: I currently have it powered with 3.3v via the programmer because it is a pain in the butt when i have to desolder it all the time to test it.

[image]

ronnieSVK, 2025-01-30:
@The-May , the "wifi" part should be involved in the code ! Only the `fast connect` should be deleted (but normally it should not cause any issue). Make sure, that after flashing is BOOT not grounded anymore, and you restarted the ESP !

cristianchelu, 2025-01-30:
@The-May, Another point you should consider if you're powering the board from the UART adapter, is if there are/how many other components may be trying to draw power from that same 3V3 line...

There might be enough juice in the programmer for the ESP chip + all others on the same 3V3 to power up and for the chip to flash, but not enough for the chip to turn on wifi (100's mA when transmitting), causing a brown-out situation.

Do you see any serial output on the `WIFITX` pad?

The-May, 2025-01-30:
I will get a proper uart and report back...takes some time. I think there is an issue with powerdraw.
I tried a semiblank esphome config and this also doesnt start so maybe the "bronout" situation might be it....
will report back....thx alot.
I will not pack everything in and maybe then the pcb will get enough power....otherwise I will try it with an uart....
Thx alot for the fast assistance by far....

cristianchelu, 2025-01-30:
I don't think there's such a thing as a "high power UART adapter" ...

Also, the amount of code in a YAML makes no difference for this.

I repeat, first, for you and for anyone that might be reading this comment in the future, this board potentially has live 240V or other high voltage electricity flowing through parts of it, be careful.

You might have to flash - plug-in - test - unplug.

If you do not want to de-solder everything -> plug device in -> test -> plug-off -> re-solder,
solder Female DuPont cables to the PCB and Male DuPonts to your UART. 

While you check stuff you can just unplug them instead of soldering.

This is why I said it's not as easy to flash this board :) in-circuit programming can be a pain sometime...

All the best,

The-May, 2025-02-08:
Update: 
New/other programmer didnt work/help.

I found the culprit:
GND and Boot0 were shorted on the chip itself - Happened before measuring with the multimeter and could not see with the bare eye that they were shorted 24/7 - i removed the shortcircuit and it works now.

[image]

Cheers guys. Thanks alot  @cristianchelu @ronnieSVK  for this journey.
I learned alot.

The-May, 2025-02-08:
Documented and Closing.

closed 2025-02-08 by The-May