dhewg/esphome-miot #35: [D][esp-idf:000]: E (3916800) system_api: Base MAC address from BLK3 of EFUSE version error, version = 56

smogger, 2024-10-17:
After the latest ESPHome 2024.10.0 update the device is no longer "accessible" entities are empty. 
Could be and ESPHome or HA issue, just trying to figure it out what is the problem. 

After update the error I saw in log is this line:
[06:45:59][D][esp-idf:000]: E (3916800) system_api: Base MAC address from BLK3 of EFUSE version error, version = 56

Whole log:
[logs_air-purifier-bedroom_logs.txt](https://github.com/user-attachments/files/17406195/logs_air-purifier-bedroom_logs.txt)

[image]

cristianchelu, 2024-10-21:
This is interesting. 

After a bit of searching I came across this mention in the esphome 2024.10 changelog:
https://esphome.io/changelog/2024.10.0.html#esp32-advanced-mac-address-configuration
> ESPHome now supports reading a custom MAC address from the ESP32 eFuse blocks and will read and use it by default if there is data burned into the eFuses. If you would like to disable this, or the eFuse has data burned that is not a MAC address into those eFuses, then you can add ignore_efuse_custom_mac: true to your ESP32 advanced configuration. This option is currently only supported when using ESP-IDF as the framework.

and this esphome issue: https://github.com/esphome/issues/issues/6333

It's not an issue in `esphome-miot`, just a breaking change in `esphome` 2024.10. Great... :)

It seems that all that's required is a new config line:
```yaml
esp32:
  # ...
  framework:
    type: esp-idf
    # ...
    advanced:
      ignore_efuse_custom_mac: true # <-------- New feature
      ignore_efuse_mac_crc: true
```

smogger, 2024-10-21:
You are godsend.... 

Issue fixed, finally it is back working again. Honestly strange since other ESP devices I have did not required this line, yet working fine. 

Now finally the devices are back and live, able to setup them again. 

Thanks

dhewg, 2025-02-08:
This should be fixed with 97f9a34495c460701f110399201d088cad4064b2, but let us know if that's not the case!

closed 2025-02-08 by dhewg