dhewg/esphome-miot #37: Smart feeder flash connection diagram [readme.md update)

jsapede, 2024-11-15:
hello,

thanks for your work ! was long time expectineg esphome conversion of smart feeder !!!

trying to flash it today. Following readme i made a diagram connection for flashing

[image]

could you confirm ? eventually include it in the readme to be sure ?

cristianchelu, 2024-11-15:
Yup. 

Double checking with the pinout of the `ESP-WROOM-02d` it looks right way around.

Also, from memory, looks pretty much the way I had it hooked up when I first flashed mine.

The wiki is open to edits, if you could add this to the `mmgg.feeder.fi1` page it'd be awesome.

Also, if you have some pictures of the disassembly process to fill in the gaps, it'd be double awesome to help others gauge the effort required. Either the short way of just removing the PCB cover or the long way of full teardown, however you managed it, is fine.

Cheers! :)

jsapede, 2024-11-16:
Ok il give a moment to add pictures 
Disassembly was quite easy, except the glue on the connectors

jsapede, 2024-11-16:
connector and soldering pads photos added

cristianchelu, 2024-11-19:
Nice

Thanks for your contribution!
I think you can close this issue now, right?

jsapede, 2024-11-19:
For sure !

dhewg, 2025-02-08:
Thanks, I noticed your diagram is already part of the wiki page!