dhewg/esphome-miot #38: pet feeder : esp_wroom_02 or esp01_1m ?

jsapede, 2024-11-16:
spent many hours today to have it work but seems ok now.

however, whent i make esphome autoconfigutation at first, it always setup an esp01_1m instead of esp_wroom_02 as specified in the config.

I erased and reinstalled all with esp_01_1m instead of esp_wroom_02 and it works (flawlessly ?).

made many try, and robably errors with the first try but never got it working with esp_wroom_02.

jsapede, 2024-11-19:
Fortunately got a second feeder AT home
 Now configs are good i'll retry with wroom

dhewg, 2025-02-08:
What's the verdict? Is that a hardware variant you discovered? Comparing chip labels may be helpful.

If a board with a smaller flash works, maybe parts of you flash are just broken? In that case, flash tools with verify steps should point that out.

In any case, if another board config works for more devices, maybe we just need to adapt our config if there's no drawback (with a comment explaining the deviation)

jsapede, 2025-02-08:
dont know exactly 
when i changes to esp01 it worked but maybe this wasnt the only change i made.
reverted back to wroom version on both feeders i have and it works perfectly

dhewg, 2025-02-08:
So let's consider it resolved, whatever `it` is/was ;)

If there're new findings just let us know!

closed 2025-02-08 by dhewg