dhewg/esphome-miot #41: Air Purifier 4 Pro ESP32 Firmware Dump

epiller, 2024-12-29:
Hey @dhewg, thanks for this amazing repo, it's been really great and works flawlessly. 
Even though this isn't an issue with your repo, I'm opening it as a public cry for help because I forgot to make an ESP32 firmware dump and now I can't go back to the original firmware in order to update the main controller's firmware.
  
### If anyone reading this has a Xiaomi Air Purifier 4 Pro and wants to help: Can you please make a dump of your ESP32 firmware before flashing ESPHome and post it here?

duckimann, 2025-02-08:
sorry i dont have a firmware dump

duckimann, 2025-02-08:
just curious, why do you have to update the main controller firmware?

duckimann, 2025-02-08:
just curious, why do you have to update the main controller firmware? @epiller

jhbruhn, 2025-02-09:
[xiaomi-air-purifier-4-pro-firmware-dump-clean.zip](https://github.com/user-attachments/files/18726157/xiaomi-air-purifier-4-pro-firmware-dump-clean.zip)

I forgot to factory-reset my purifier before doing this dump, so I removed my wifi credentials manually with a hex-editor. Hopefully I didn't break anything, but I guess this is worth a shot!

dhewg, 2025-02-14:
Nice, that should settle it!

@epiller It's not clear why you want/need to update the MCU firmware. If there's a problem with one version, which is fixed by a newer one: That would be a relevant information for other users here as well: so please share any insights if that's the case!

closed 2025-02-14 by dhewg

epiller, 2025-03-04:
@jhbruhn Thank you so much!
@dhewg The Level and motor RPM are updated too infrequently, which I'm hoping is something that will be fixed in a newer firmware. Thanks to @jhbruhn, I can revert back to the stock firmware, update the main MCU, and go back to your firmware.

cristianchelu, 2025-03-04:
> [@jhbruhn](https://github.com/jhbruhn) Thank you so much! [@dhewg](https://github.com/dhewg) The Level and motor RPM are updated too infrequently, which I'm hoping is something that will be fixed in [...]

For what it's worth, I had success with manually queuing the update command, try to see if this works, replacing `get_properties 9 1` with your actual motor speed SIID and PIID numbers:
```yaml
time:
  - platform: homeassistant
    on_time:
      - seconds: /1 # every 1 second, for example
        then:
          - lambda: id(miot_main).queue_command("get_properties 9 1");
```

More info: https://esphome.io/components/time/index.html#on-time-trigger

I use mine to approximate the active power on-device instead of using Powercalc integration (a few power metering smart socket measurements at different speeds + online exponential fit calculator for the numbers):
```yaml
- platform: "template"
    name: "Active power"
    id: active_power
    state_class: measurement
    unit_of_measurement: W
    icon: mdi:flash
    accuracy_decimals: 2
    lambda: |-
      float speed = id(motor_speed).state;
      float anion = id(ionizer).state ? 1.0 : 0.0;
      float fan = 1.17067 * std::pow(2.71828, 0.00157301 * speed);
      return fan + anion;
    update_interval: 1s
    on_value: 
      then:
        lambda: 'id(miot_main).queue_command("get_properties 9 1");'
```

cristianchelu, 2025-03-05:
@epiller 
> Did any of you manage to configure your ESPHome code to expose the purifier as a fan with speed and modes?

There's a WIP branch that has fan component support, see [here](https://github.com/dhewg/esphome-miot/issues/19#issuecomment-2653593134)

To use it change the external component branch
```yaml
external_components:
  source: github://dhewg/esphome-miot@fan
```

Or if that's not enough customization for you and you have patience, I made a hacky but infinitely customizable initial implementation using the template fan component. Search for `template_fan` in [this config](https://github.com/dhewg/esphome-miot/blob/main/config/dmaker.fan.p18.yaml) to get an initial idea of what's required. 

More details and chatter in the same thread from above starting from [here](https://github.com/dhewg/esphome-miot/issues/19#issuecomment-2241180251), if you haven't already read it.

> As it stands, too many individual commands need to be exposed to HomeKit in order for it to be usable over it (power, ionizer, speed readout, mode, manual speed, possibly even fan level).
> I tried multiple times with template fan but failed miserably...

Depending on what you mean by "failed", I hope the config and thread I linked above will help.

epiller, 2025-03-06:
@cristianchelu The thread just keeps on giving! Again, thank you! I opted for the fan branch and it works almost flawlessly. I do have a few suggestions if you're open to it:  
- It would be nice if, when turned on, the fan went to the last preset mode instead of switching to manual. 
- It'd be useful to show the current fan level when a preset mode is active. Manually adjusting the level while a preset is active would set it to "Manual", as it does right now.
- When exposed to HomeKit, it's shown as a combo of 3 switches and a fan (screenshot below). I remember IKEA Starkvind (via ZHA) had this problem for some time but I they managed to solve it.

[image]

dhewg, 2025-03-06:
FYI: I just merged the fan branch

> * It would be nice if, when turned on, the fan went to the last preset mode instead of switching to manual.

There's a workaround on the main branch for what I believe is a esphome bug, which solves this. See the linked issue above for the details.

> * It'd be useful to show the current fan level when a preset mode is active. Manually adjusting the level while a preset is active would set it to "Manual", as it does right now.

From what I can see a fan component is not designed to provide that. Again, see the linked issue above for all the details about this.

> * When exposed to HomeKit, it's shown as a combo of 3 switches and a fan (screenshot below). I remember IKEA Starkvind (via ZHA) had this problem for some time but they managed to solve it somehow.

AFAICT it works as expected in HA, so no idea about HomeKit, but let us know if there's anything we can do about here.