dhewg/esphome-miot #42: New devices quickstart utility

cristianchelu, 2025-02-06:
Heya,

I've just released a utility app that may be of use to anyone trying to reverse engineer new devices.

It works by parsing the JSON spec file and trying to map each service/property/event to the most likely esphome component type, then generating an ESPHome YAML config using this custom component at its core.

That config should have all the proper SIIDs/PIIDs laid out from the start, easing the frustration of starting a new config.

All thoughts welcome.

PS: I figured it would better live as a separate entity in its own repo, but if you want to have it close to this component we can figure out a way to merge it here.

[image]
[image]

dhewg, 2025-02-07:
That looks great, nice work!

And I agree on it living in a seperate repo. If you want, you could add a link to the README here and I'll happily merge that ;)

helgek, 2025-02-07:
@cristianchelu Looks very interesting, thank you! 

I tried the search but it didn't turn any results. Tried e.g.:

fan:0000A005:dmaker-p18:1
Smart
Fan
etc.

but no luck.

cristianchelu, 2025-02-07:
> @cristianchelu Looks very interesting, thank you! 
> 
> I tried the search but it didn't turn any results. Tried e.g.:
> 
> fan:0000A005:dmaker-p18:1
> Smart
> Fan
> etc.
> 
> but no luck. 

Do you have any errors in the console? It should work exactly like the search page on the miot website because that's what it's calling in the background

helgek, 2025-02-07:
@cristianchelu the only response I see is a short animation of the search icon next to the search field. Apart from this I don't see any other activity on the page

closed 2025-02-08 by dhewg

cristianchelu, 2025-02-08:
> @cristianchelu the only response I see is a short animation of the search icon next to the search field. Apart from this I don't see any other activity on the page. And I tried 3 different browsers.

I will add code that shows any errors in the UI, but in the meantime 

@helgek please open an issue on https://github.com/cristianchelu/miot2esphome and let's continue there so we don't ping everyone else subscribed to this repo.

if you could open the developer tools console of your browser ([here is a random article](https://help.frontlineeducation.com/hc/en-us/articles/35372818850323-How-to-Access-the-Browser-Console) I found that describes the steps for all browsers)

then *with the console open* try to search, and copy/paste any error messages (red background) that appear there to the new issue, I'd be grateful!

Thanks!

helgek, 2025-02-09:
@cristianchelu Thank you for the hint with the console in the developer tools. I wasn't aware of it. There was this error:

```
corsproxy.io/?url=https%3A%2F%2Fhome.miot-spec.com%2Fs%2Ffan:1         
       Failed to load resource: net::ERR_CONNECTION_RESET
```

but after disabling the company VPN connection the search worked. So I guess it was related to some network issue on my side/firewall setting of the company/filter setting for that url or something else into this direction. Sorry for bringing this up, your tool seems to be alright!

PR #43 (2025-02-08): Readme changes
Per your suggestion, I've added a link in the readme to the miot2esphome utility. 
Thanks!

Closes #42, I think. :)

I also let a bit of ADHD take over and added some more detail to the unsupported devices section, based on what I saw in the issues sections.
It's in a different commit, let me know if I should remove it from this PR.

Also, feel free to make any text changes, I didn't put too much thought into writing this.