dhewg/esphome-miot #44: Any tutoriel for petfeeder ?

Idaho947, 2025-02-09:
Hello,

I try to flash my xiaomi pet feeder but block here : Using the tool of your choosing (esptool recommended) download the contents of the flash and store for safe keeping, in case you want to revert to factory or sell the device.

Can someone help me tondo that ? I can flash the board  in ha in esphome but don t understand how to flash the wright files.

The-May, 2025-02-10:
As far as i know you cannot dump/backup with esphome.
Therefore you explicitly need esptool.
These are the basic commands
https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html

see read flash content
https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html

So first you need to find out which esp-chip is there and how much MB it has. It could be 2 or 4 or something else.
so after setting up your python venv for esptool you run:
`esptool.py flash_id`

See example:

[image]

after that command you get how much MB it has. (IT CAN BE DIFFERENT FOR YOU)

and then you run:
esptool.py read_flash 0x000000 [insert_size_of_flash] petfeeder_backup.bin

so for example if it is 4mb
esptool.py read_flash 0x000000 0x400000 petfeeder_backup.bin

hint: if your esptool.py command is stuck at 25/50/75% it might be possible that you wrote a wrong mb-number
 

alternatively you could try using the  Flash download tool, but i realized the dumped .bin seems to be different compared to esptool, so I dont really recommend it since I dont know if this is good.
https://www.espressif.com/en/support/download/other-tools

Idaho947, 2025-02-10:
Thank for your reply ! And it's work well for me. Now I have to put the espcard in the feeder. Hope it works !

Idaho947, 2025-02-13:
Ok nice it works very well ! My rabbit thank you. They don't have to wait for their week end morning meel !

closed 2025-02-13 by Idaho947