dhewg/esphome-miot #47: Air Purifier 3C (zhimi.airpurifier.mb4): EFUSE CRC error

The-May, 2025-02-10:
I recently tried to flash my zhimi.airpurifier.mb4 and It seems to have issues on booting with the current code from 
https://github.com/dhewg/esphome-miot/blob/main/config/zhimi.airp.mb4a.yaml

i used 1:1 the code from it, changed some secret vars (because mine were a lil different, but they work in other projects!)

I followed the wiki https://github.com/dhewg/esphome-miot/wiki/Xiaomi-Mi-Air-Purifier-3C

It seems to be that the code might not be totally suitable for the purifier......

see logs when connecting to the logs via usb:

```

`[C][wifi:062]:   Local MAC: re:da:ct:ed:00:00
[D][esp-idf:000][wifi]: I�ets Jun  8 2016 00:22:57
[18:24:34]
[18:24:34]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:24:34]configsip: 0, SPIWP:0xee
[18:24:34]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:24:34]mode:DIO, clock div:2
[18:24:34]load:0x3fff0030,len:7104
[18:24:34]load:0x40078000,len:15516
[18:24:34]load:0x40080400,len:4
[18:24:34]load:0x40080404,len:3668
[18:24:34]entry 0x4008061c
[18:24:34]I (29) boot: ESP-IDF 5.1.5 2nd stage bootloader
[18:24:34]I (29) boot: compile time Feb 10 2025 18:17:24
[18:24:34]W (29) boot: Unicore bootloader
[18:24:34]I (33) boot: chip revision: v1.0
[18:24:34]I (36) boot.esp32: SPI Speed      : 40MHz
[18:24:34]I (41) boot.esp32: SPI Mode       : DIO
[18:24:34]I (46) boot.esp32: SPI Flash Size : 4MB
[18:24:34]I (50) boot: Enabling RNG early entropy source...
[18:24:34]I (56) boot: Partition Table:
[18:24:34]I (59) boot: ## Label            Usage          Type ST Offset   Length
[18:24:34]I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
[18:24:34]I (74) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[18:24:34]I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
[18:24:34]I (89) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[18:24:34]I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[18:24:34]I (104) boot: End of partition table
[18:24:34]I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2d6d8h (186072) map
[18:24:34]I (184) esp_image: segment 1: paddr=0003d700 vaddr=3ffb0000 size=02918h ( 10520) load
[18:24:34]I (188) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=9bf78h (638840) map
[18:24:35]I (420) esp_image: segment 3: paddr=000dbfa0 vaddr=3ffb2918 size=00838h (  2104) load
[18:24:35]I (421) esp_image: segment 4: paddr=000dc7e0 vaddr=40080000 size=14a44h ( 84548) load
[18:24:35]I (471) boot: Loaded app from partition at offset 0x10000
[18:24:35]I (471) boot: Disabling RNG early entropy source...
[18:24:35]I (483) cpu_start: Unicore app
[18:24:35]I (484) cpu_start: Pro cpu up.
[18:24:35]I (484) cpu_start: Single core mode
[18:24:35]I (495) cpu_start: Pro cpu start user code
[18:24:35]I (495) cpu_start: cpu freq: 160000000 Hz
[18:24:35]I (495) cpu_start: Application information:
[18:24:35]I (500) cpu_start: Project name:     testtest
[18:24:35]I (505) cpu_start: App version:      2024.12.4
[18:24:35]I (510) cpu_start: Compile time:     Feb 10 2025 18:16:31
[18:24:35]I (516) cpu_start: ELF file SHA256:  a13d7a14ebf7e99d...
[18:24:35]I (522) cpu_start: ESP-IDF:          5.1.5
[18:24:35]I (527) cpu_start: Min chip rev:     v0.0
[18:24:35]I (532) cpu_start: Max chip rev:     v3.99 
[18:24:35]I (537) cpu_start: Chip rev:         v1.0
[18:24:35]I (541) heap_init: Initializing. RAM available for dynamic allocation:
[18:24:35]I (549) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[18:24:35]I (555) heap_init: At 3FFB73F0 len 00028C10 (163 KiB): DRAM
[18:24:35]I (561) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[18:24:35]I (567) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
[18:24:35]I (573) heap_init: At 40094A44 len 0000B5BC (45 KiB): IRAM
[18:24:35]I (579) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
[18:24:35]I (587) spi_flash: detected chip: generic
[18:24:35]I (590) spi_flash: flash io: dio
[18:24:35]I (595) app_start: Starting scheduler on CPU0
[18:24:35]I (599) main_task: Started on CPU0
[18:24:35]I (603) main_task: Calling app_main()
[18:24:35]0;32m[I][logger:171]: Log initialized
[C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][wifi:048]: Setting up WiFi...
[D][esp-idf:000]: I (715) wifi:
[D][esp-idf:000]: wifi driver task: 3ffc26a8, prio:23, stack:6656, core=0
[D][esp-idf:000]: 

[D][esp-idf:000][wifi]: E (729) system_api: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0xee; calc_crc = 0xd5

[D][esp-idf:000][wifi]: W (738) system_api: Ignore MAC CRC error

[D][esp-idf:000][wifi]: I (753) wifi:
[D][esp-idf:000][wifi]: wifi firmware version: bbd58fc31
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (769) wifi:
[D][esp-idf:000][wifi]: wifi certification version: v7.0
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (779) wifi:
[D][esp-idf:000][wifi]: config NVS flash: enabled
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (800) wifi:
[D][esp-idf:000][wifi]: config nano formating: disabled
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (821) wifi:
[D][esp-idf:000][wifi]: Init data frame dynamic rx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (831) wifi:
[D][esp-idf:000][wifi]: Init static rx mgmt buffer num: 5
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (851) wifi:
[D][esp-idf:000][wifi]: Init management short buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (871) wifi:
[D][esp-idf:000][wifi]: Init dynamic tx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (882) wifi:
[D][esp-idf:000][wifi]: Init static rx buffer size: 1600
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (902) wifi:
[D][esp-idf:000][wifi]: Init static rx buffer num: 10
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (923) wifi:
[D][esp-idf:000][wifi]: Init dynamic rx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000]: I (934) wifi_init: rx ba win: 6

[D][esp-idf:000]: I (943) wifi_init: tcpip mbox: 32

[D][esp-idf:000]: I (953) wifi_init: udp mbox: 6

[D][esp-idf:000]: I (954) wifi_init: tcp mbox: 6

[D][esp-idf:000]: I (954) wifi_init: tcp tx win: 5760

[D][esp-idf:000]: I (974) wifi_init: tcp rx win: 5760

[D][esp-idf:000]: I (974) wifi_init: tcp mss: 1440

[D][esp-idf:000]: I (984) wifi_init: WiFi IRAM OP enabled

[D][esp-idf:000]: I (984) wifi_init: WiFi RX IRAM OP enabled

[C][wifi:061]: Starting WiFi...
[D][esp-idf:000]: E (995) system_api: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0xee; calc_crc = 0xd5

[D][esp-idf:000]: W (1005) system_api: Ignore MAC CRC error

[C][wifi:062]:   Local MAC: re:da:ct:ed:00:0`
```

Is it just me or can someone recreate the same issue I have with the same model?

The-May, 2025-02-10:
> Is your physical chip labelled `ESP-WROOM-32D` or something else?
> 
> Does your esp32 block include the unicore sdk config option, as per the yaml you linked?
> 
> esp32:
>   board: esp32dev
>   framework:
>     type: esp-idf
>     sdkconfig_options:
>       CONFIG_FREERTOS_UNICORE: y # <-- This is what the error says is missing
>     advanced:
>       ignore_efuse_custom_mac: true
>       ignore_efuse_mac_crc: true

yes the chip clearly says esp-wroom-32d, see picture i just made

[image]

i tried both esp32doit and esp32dev (for debugging/testing)

[image]

It is now 

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_custom_mac: true
      ignore_efuse_mac_crc: true
```

just like in the original yaml i have copied.
I also tried CONFIG_FREERTOS_UNICORE: "y" without any success.

The-May, 2025-02-10:
> Works for me too, I guess there's something wrong with your environment where you build in:
> 
> ```
> $ esphome version 
> Version: 2024.12.4
> 
> $ grep -w board -A 7 config/purifier.yaml      
>   board: esp32dev
>   framework:
>     type: esp-idf
>     sdkconfig_options:
>       CONFIG_FREERTOS_UNICORE: y
>     advanced:
>       ignore_efuse_custom_mac: true
>       ignore_efuse_mac_crc: true
> 
> $ esphome compile config/purifier.yaml
> [...]
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
>  - framework-espidf @ 3.50105.0 (5.1.5) 
>  - tool-cmake @ 3.21.3 
>  - tool-esptoolpy @ 4.8.1 
>  - tool-mklittlefs @ 3.2.0 
>  - tool-ninja @ 1.7.1 
>  - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
>  - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
>  - toolchain-esp32ulp @ 2.35.0-20220830 
>  - toolchain-xtensa-esp32 @ 12.2.0+20230208
> [...]
> 
> $ grep CONFIG_FREERTOS_UNICORE config/.esphome/build/purifier/sdkconfig.* 
> config/.esphome/build/purifier/sdkconfig.purifier:CONFIG_FREERTOS_UNICORE=y
> ```

Im using the ESPhome Device Builder Addon with gui and so.

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/esp32d-mipurifier01-4fca99.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing purifier (board: esp32doit-devkit-v1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
Compiling .pioenvs/purifier/src/main.cpp.o
Linking .pioenvs/purifier/firmware.elf
RAM:   [=         ]   9.1% (used 29920 bytes from 327680 bytes)
Flash: [=====     ]  53.1% (used 974789 bytes from 1835008 bytes)
Building .pioenvs/purifier/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/purifier/firmware.bin"], [".pioenvs/purifier/firmware.elf"])
Wrote 0xff970 bytes to file /data/build/purifier/.pioenvs/purifier/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/purifier/firmware.bin"], [".pioenvs/purifier/firmware.elf"])
========================= [SUCCESS] Took 19.05 seconds =========================
INFO Successfully compiled program.
```
Unfortunately I cant browse/find the dir where the build is. -> I think it is nowhere in /config/ or on those other shares......
I can try and create a docker-container if needed, but looking @ the versions they look the same....

The-May, 2025-02-11:
Update: created a venv on Windows with latest esphome, same issue when i watch the logs on the ESP:

Steps I did:
created venv etc.
added config and secrets.yaml and purifier.yaml

then did `(venvesphome) C:\Users\May10_v2\Desktop\esphome>esphome compile config/purifier.yaml`

compiling took some time becasue it starts from absolute scratch.
It gave me 
`Wrote 0xff990 bytes to file C:\Users\May10_v2\Desktop\esphome\config\.esphome\build\purifier\.pioenvs\purifier/firmware.factory.bin, ready to flash to offset 0x0`
I flashed with ESPHOME web flasher https://web.esphome.io/ (just in case mine is broken/faulty)

in the meantime i checked if CONFIG_FREERTOS_UNICORE equals y in the dir you asked me to check:

[image]

Are you guys @dhewg  @cristianchelu  using zhimi.airpurifier.mb4 or zhimi.airp.mb4a?
I can also send you my .bin (without any critical secrets) to cross out the issue....or send me your bin without any critical credentials so we can find out if this is an issue related to my purifier revision maybe.....

Does the external_components:
  source: github://dhewg/esphome-miot@main maybe overwrite something, so CONFIG_FREERTOS_UNICORE gets defaulted to n or so? It feels like im missing something out.

Can also retry on Docker if needed, but would take a lil bit more time though....

The-May, 2025-02-11:
It looks like the ESP-Chip is somehow very strange.
Using the espdownloadertool and reading the chip info gives me following:

```
Chip is unknown ESP32 (revision v1.0)
Features: WiFi, BT, **Single Core**, 240MHz, VRef calibration in efuse, Coding Scheme 3/4
Crystal is 40MHz
MAC: redacted
Manufacturer: 1c
Device: 7016
Status value: 0xffff00
Detected flash size: 4MB
```

But looking at the Datasheets on https://www.espressif.com/en/products/socs it should be dual core..

this should be for: ESP32-WROOM-32D :
[ESP32-WROOM-32D](https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32d_esp32-wroom-32u_datasheet_en.pdf)

and looking at the data, the esp uses 2 cores, so something is beyond standard and beyond my imagination to try to be honest.

I even did a "first time" installation brings me to the "you only have 1 core hurrdurr" issue. It feels like esphome totally ignores my code to use one core.

dhewg, 2025-02-11:
I replied earlier, but it looks like that got lost...

Anyway, are you sure you're flashing the correct file?
You can use esphome for that too, fwiw.
It's been a while since I did that myself, but something like:
esphome run --device /dev/ttyUSB0 config/purifier.yaml

The-May, 2025-02-12:
> I replied earlier, but it looks like that got lost...
> 
> Anyway, are you sure you're flashing the correct file?
> You can use esphome for that too, fwiw.
> It's been a while since I did that myself, but something like:
> esphome run --device /dev/ttyUSB0 config/purifier.yaml

I am still currently trying to get it to work and looked up many other "related" issues and I managed to get network connectivity (wifi) working when using.
What I already can say and found out:
1.
```
esp32:
  board: esp32doit-devkit-v1
```
seems to be working more reliable since it is singe core by default.

2. clearing the build when code has been done, seems to fix some issues (it looks like some changes dont get applied)

when using the esp32diot-devkit-v1 board, i get  mac crc-errors but according to someone (can no longer find the link) they are related because mac-adress is "written" in the efuse and since the device does not have enough power (brownout), the fuse somehow is temporarily "altered".
After hooking it up to main power it seems to boot without issues and i can already ping, but I am still getting some pesky cosmetic issues. 
Will report when I found out what the issue might be. My brain is melting and fuming :)

cristianchelu, 2025-02-12:
@The-May 
> when using the esp32diot-devkit-v1 board, i get mac crc-errors but according to someone (can no longer find the link) they are related because mac-adress is "written" in the efuse and since the device does not have enough power (brownout), the fuse somehow is temporarily "altered".
Just a quick note related only to the efuse.

Xiaomi writes unknown non-mac data to the efuses (they're allowed to) on many devices.
Throw this advanced block in the esp32 section and you should be good on this front:
```yaml
esp32:
  # board: whatever
  framework:
    type: esp-idf
    # other stuff
    advanced:
      ignore_efuse_custom_mac: true # <- this
      ignore_efuse_mac_crc: true # <- and this
```
I'm at a loss for comments on your other issues right now, unfortunately.

The-May, 2025-02-12:
> [@The-May](https://github.com/The-May)
> 
> > when using the esp32diot-devkit-v1 board, i get mac crc-errors but according to someone (can no longer find the link) they are related because mac-adress is "written" in the efuse and since the device does not have enough power (brownout), the fuse somehow is temporarily "altered".
> > Just a quick note related only to the efuse.
> 
> Xiaomi writes unknown non-mac data to the efuses (they're allowed to) on many devices. Throw this advanced block in the esp32 section and you should be good on this front:
> 
> esp32:
>   # board: whatever
>   framework:
>     type: esp-idf
>     # other stuff
>     advanced:
>       ignore_efuse_custom_mac: true # <- this
>       ignore_efuse_mac_crc: true # <- and this
> 
> I'm at a loss for comments on your other issues right now, unfortunately.

Thanks for pointing that out.
But im always using this

```
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_custom_mac: true
      ignore_efuse_mac_crc: true
```
which gives me the most reliable results.
pinging it was also possible at some point

I am currently trying to reproduce my steps (from 0) so i can document it on the wiki, so at least the steps can be redone reliably if someone has the same poopbox like me

dhewg, 2025-02-12:
You're making it too hard yourself :P

Don't change too many things at the same time, and I recommend to not 
mix tools. A esphome venv is really all you need.

Of course you should also always use the device's original power source 
for runtime testing. Anything else is only temporarily for e.g. flashing.

A vendor can use the esp fuses as desired. That's where the mac crc 
mismatch comes from, as well as the single core issue:

efuse_summary:
DISABLE_APP_CPU (BLOCK0)                           Disables APP CPU 
                              = True R/W (0b1)
[...]
MAC (BLOCK0)                                       MAC address 

    = [redacted] (CRC 0x8e invalid - calculated 0x31) R/W
MAC_CRC (BLOCK0)                                   CRC8 for MAC address 
                              = 142 R/W (0x8e)

So they chose to enable DISABLE_APP_CPU, which is where our 
CONFIG_FREERTOS_UNICORE comes from.
And they chose to not respect the crc8 for the mac, which is where our 
ignore_efuse_mac_crc comes from.
Removing any of those config options will only yield problems.

So get a stable setup first, and the yaml in config/ should just work 
just fine. Everything else is fine tuning, where ota updates are all you 
need. You really have to try hard to brick it (as in: only reflashing a 
working firmware using bootloader mode fixes it)

The-May, 2025-02-12:
SUCCESS 🥇 

So what I did:
1. tried to compile and upload with esphome builder (Home assistant Addon)  -> CONFIG_FREERTOS_UNICORE  or Fuse-Issues -> no wifi
2. created venv on withdows, compiled via cli and uploaded with esphome (web) (iirc) -> CONFIG_FREERTOS_UNICORE  or Fuse-Issues -> no wifi
3.created venv on linux (latest ubuntu machine), compiled via cli (on a 2nd gen i5 ~ 5minutes) and uploaded firmware.factory.bin with esphome (web) -> success.
I am really baffled why this works to be honest. I did the same, used the same adapter and since step 2 and 3 SHOULD be the same I cannot fathom why it now worked.

Do you guys @dhewg @cristianchelu know why there could be a "missmatch" in handling?
Or was I just stupid and did everything totally different or are there different handlings between these enviroments?

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/esp32d-mipurifier01-xxxxx.yaml...
INFO Starting log output from 192.168.1.33 using esphome API
INFO Successfully connected to esp32d-mipurifier01-xxxxx @ 192.168.1.33 in 0.056s
INFO Successful handshake with esp32d-mipurifier01-xxxxx @ 192.168.1.33 in 0.075s
[14:31:27][I][app:100]: ESPHome version 2024.12.4 compiled on Feb 12 2025, 14:20:08
[14:31:27][I][app:102]: Project dhewg.esphome-miot version zhimi.airp.mb4a
[14:31:27][C][wifi:600]: WiFi:
[14:31:27][D][esp-idf:000]: E (484885) system_api: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0xee; calc_crc = 0xd5

[14:31:27][D][esp-idf:000]: W (484890) system_api: Ignore MAC CRC error

[14:31:27][C][wifi:428]:   Local MAC: [redacted]
[14:31:27][C][wifi:433]:   SSID: [redacted]
[14:31:27][C][wifi:436]:   IP Address: 192.168.1.33
[14:31:27][C][wifi:439]:   BSSID: [redacted]
[14:31:27][C][wifi:441]:   Hostname: 'esp32d-mipurifier01-xxxxx'
[14:31:27][C][wifi:443]:   Signal strength: -36 dB ▂▄▆█
[14:31:27][C][wifi:447]:   Channel: 11
[14:31:27][C][wifi:448]:   Subnet: 255.255.255.0
[14:31:27][C][wifi:449]:   Gateway: 192.168.1.1
[14:31:27][C][wifi:450]:   DNS1: 192.168.1.1
[14:31:27][C][wifi:451]:   DNS2: 0.0.0.0
[14:31:27][I][miot:183]: Network status changed to 'cloud'
[14:31:27][D][miot:163]: Queuing MCU command 'MIIO_net_change cloud'
[14:31:27][C][logger:185]: Logger:
[14:31:27][C][logger:186]:   Level: DEBUG
[14:31:27][C][logger:188]:   Log Baud Rate: 115200
[14:31:27][C][logger:189]:   Hardware UART: UART0
[14:31:27][C][uart.idf:159]: UART Bus 1:
[14:31:27][C][uart.idf:160]:   TX Pin: GPIO17
[14:31:27][C][uart.idf:161]:   RX Pin: GPIO16
[14:31:27][C][uart.idf:163]:   RX Buffer Size: 256
[14:31:27][C][uart.idf:165]:   Baud Rate: 115200 baud
[14:31:27][C][uart.idf:166]:   Data Bits: 8
[14:31:27][C][uart.idf:167]:   Parity: NONE
[14:31:27][C][uart.idf:168]:   Stop bits: 1
[14:31:27][C][miot:132]: MIoT:
[14:31:27][C][miot:140]:   Time: UNAVAILABLE
[14:31:27][C][miot.switch:068]: MIoT Switch 'Power'
[14:31:27][C][miot.switch:070]:   Icon: 'mdi:power'
[14:31:27][C][miot.switch:090]:   Restore Mode: always OFF
[14:31:27][C][miot.switch:025]:   SIID: 2
[14:31:27][C][miot.switch:026]:   PIID: 1
[14:31:27][C][miot.switch:068]: MIoT Switch 'Notification Sounds'
[14:31:27][C][miot.switch:070]:   Icon: 'mdi:volume-high'
[14:31:27][C][miot.switch:090]:   Restore Mode: always OFF
[14:31:27][C][miot.switch:025]:   SIID: 6
[14:31:27][C][miot.switch:026]:   PIID: 1
[14:31:27][C][miot.switch:068]: MIoT Switch 'Child Lock'
[14:31:27][C][miot.switch:070]:   Icon: 'mdi:lock'
[14:31:27][C][miot.switch:090]:   Restore Mode: always OFF
[14:31:27][C][miot.switch:025]:   SIID: 8
[14:31:27][C][miot.switch:026]:   PIID: 1
[14:31:27][C][miot.select:040]: MIoT Select 'Mode'
[14:31:27][C][miot.select:041]:   SIID: 2
[14:31:27][C][miot.select:042]:   PIID: 4
[14:31:27][C][miot.select:043]:   Options are:
[... 50 lines cut ...]
[14:31:27][C][miot.button:012]:   AIID: 1
[14:31:27][C][miot.button:013]:   args: 3 0
[14:31:27][C][captive_portal:089]: Captive Portal:
[14:31:27][C][mdns:116]: mDNS:
[14:31:27][C][mdns:117]:   Hostname: esp32d-mipurifier01-xxxxx
[14:31:27][C][esphome.ota:073]: Over-The-Air updates:
[14:31:27][C][esphome.ota:074]:   Address: esp32d-mipurifier01-xxxxx.local:3232
[14:31:27][C][esphome.ota:075]:   Version: 2
[14:31:27][C][esphome.ota:078]:   Password configured
[14:31:27][C][safe_mode:018]: Safe Mode:
[14:31:27][C][safe_mode:019]:   Boot considered successful after 60 seconds
[14:31:27][C][safe_mode:021]:   Invoke after 10 boot attempts
[14:31:27][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[14:31:27][C][api:140]: API Server:
[14:31:27][C][api:141]:   Address: esp32d-mipurifier01-xxxxx.local:6053
[14:31:27][C][api:143]:   Using noise encryption: YES
[14:31:55][W][miot:346]: MCU time request: no time source available
[14:31:55][D][miot:381]: MCU time request: 0
```

Logs also look good.
This took me an eternity and I will def. documentate what to do. step by step...
At least my purifier now has a subtle interface to flash:

[image]

dhewg, 2025-02-12:
Nice, progress at last!

But honestly: no idea where to start looking, the whole stack is rather 
big, the issue could be in esphome, in platformio, in the esp toolchain 
or one of the metric ton of tools used by each.

If you care, you could try a minimal esphome project without these 
components here, to illustrate that CONFIG_FREERTOS_UNICORE gets somehow 
ignored on windows and report that to esphome.

The-May, 2025-02-12:
documentated working steps

closed 2025-02-12 by The-May

sebPomme, 2025-02-27:
@The-May thanks you for the idea of `a subtle interface to flash` ✨