dhewg/esphome-miot #50: https://home.miot-spec.com/spec?type=urn:miot-spec-v2:device:fan:0000A005:dmaker-02:1

d0np3p3, 2025-02-26:
I got some of the Dreammaker fans, they use their own server.
Could we get them flashed and work with esphome? 

There are 2 Versions: 
Dream Maker Smart Fan DM-FAN01
Dream Maker Smart Fan DM-FAN02-W (battery-powered version)

Some Information: 
https://home.miot-spec.com/spec?type=urn:miot-spec-v2:device:fan:0000A005:dmaker-02:1
https://community.home-assistant.io/t/dreammaker-feel-fan-plus-dream-maker-dm-fan02-w/319885

[image]

dhewg, 2025-02-28:
Maybe, if that chip in the middle reading "ARM" is the mcu.

But it should be easy to find out:
Wire your 3.3v uart ttl to the TX/RX/GND labeled pads on the bottom left and fetch the serial output while booting up the board. That log of the vendor firmware should be enough to find out if this board is supported.

dhewg, 2025-02-28:
That looks different to what my purifier prints. While I do see the raw serial communication, here it's just mentioned, but it's not clear if it's really the miot serial protocol, which is what we need.

So I think you need to try yourself and check what the esp32 gets from the mcu over serial.
For that you have two options:
1) Don't touch the fw and sniff the serial communication between the two chips
2) Flash a bare esphome fw and check

All esp32 configs here have tx_pin=17 and rx_pin=16, so that's what I would try here first for either option.

For 2), make sure flashing via bootloader works, and use that to make a fw backup first.

d0np3p3, 2025-02-28:
I am trying esptool right now, maybe you got a hint how to sniff the serial communication?

dhewg, 2025-03-01:
If you're careful and have the tools it should be relatively easy to 
directly attach to the esp32's pins.

See the pin layout on page 8 here:
https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32d_esp32-wroom-32u_datasheet_en.pdf

As mentioned before, the esp is likely using pin 16 and 17 as a serial 
line to the mcu. So IO16 and IO17, which is UART2.

d0np3p3, 2025-03-05:
on IO16 i only get some strings like "▒" with baudrate 115200 or"_YUAAX▒▒XUAA▒Ua▒" with 9600 
any hints? 
I Use Putty with a  usb ttl

dhewg, 2025-03-05:
Hard to tell, maybe the serial between the two chips use rts/cts and you 
can't sniff it wired like you have atm.

I'd probably just flash a bare esphome config with "level: VERBOSE" to 
see what the comms look like. But of course you'd have to make sure you 
have a proper fw backup to go back to in case the device is imcompatible.

d0np3p3, 2025-04-03:
I tried to backup the fw multiple times, mostly the esp reboots, the few times it worked i get different files, so i don't dare to flash esphome. 
I will give it another try if i have more time.
Maybe anybody else with a dreammaker will give it a try.

d0np3p3, 2025-04-17:
I think this fans are not using the Xiaomi Miot Serial Communication, so i gave up right now.
As workaround i use the [dmiot2mqqt](https://github.com/rapejim/dmiot2mqtt)
If someone has done some more research, we could reopen the issue

closed 2025-04-17 by d0np3p3

BobeOlsen, 2026-05-23:
> Thank you, i got it with your findings, i didn't get output from mcu. Esphome runs and working, some small issues to fix,and test long running. See my repo.

Glad the info was useful that means I actually did something that works
Same here works fine with homekit also few tweaks needed

BobeOlsen, 2026-05-24:
> [@BobeOlsen](https://github.com/BobeOlsen) Are you going to share your project?

You mean the code/sketch? 
I dont want to deal with all the github way of doing it like repo or whatever its called
I can just paste arduino sketch here and if it will help somebody else that would be nice