dhewg/esphome-miot #54: Support for zhimi.airp.mb5a?

grimmaldus, 2025-04-29:
Hi and thank you for this awesome project!

I noticed that `zhimi.airp.mb5` is listed in the supported devices. I have a device labeled `zhimi.airp.mb5a`, which seems to be a close variant.

- Has anyone reported success using it with this component?
- Do you know if it's likely to work or if testing/adjustments might be needed?

I’d be happy to provide any close information or help test if needed.

Thanks again!

cristianchelu, 2025-04-29:
Hey @grimmaldus, 

As far as I'm aware, you'd be the first one in this project's discussions trying out `mb5a`.

The published MIoT specs for both devices are identical, and we have confirmation that the same yaml worked for `mb4` and `mb4a`, so it's worth a shot.

If your device has the same disassembly procedure as my `mb5`, it's a 10 minute process, check its [wiki page](https://github.com/dhewg/esphome-miot/wiki/Xiaomi-Smart-Air-Purifier-4).

If you can, take notes/pictures if you see any differences:
- in the dissassembly process
- in the PCB / Wifi chip (Mine is `ESP-WROOM-32D`)

and add them to the wiki page.

Just remember to take a backup of the original firmware before flashing, just in case.

Let us know how it goes!

grimmaldus, 2025-05-01:
@cristianchelu Thanks for the detailed reply!

I just started disassembling the zhimi.airp.mb5a and so far everything looks more or less the same as your mb5. I’ve taken some photos of the board, and while the PCB layout is slightly different and marked as a different revision (REV...), the most important part—the ESP module—is identical: it’s an ESP-WROOM-32D, just like yours.

I’ll confirm that the flashing pads match, and then proceed with backing up the original firmware before flashing the zhimi.airp.mb5 YAML using ESPHome.

I’ll report back with results, and I’ll also upload the pictures and note any relevant differences on the wiki if everything works out.

Thanks again for the guidance!

[image]

[image]

grimmaldus, 2025-05-01:
@cristianchelu 

> Did you power it on via the UART programmer in the above logs?
>It might be that they usually don't supply enough energy to properly boot the ESP32 and whatever else is sharing that same 3V3 bus.
>If yes, re-assemble it partially so everything is connected and safe and try a normal power on from the wall, see if you get anything over wifi.

I tried powering it normally (from the wall) with everything reassembled. The LED screen shows a Wi-Fi error (a "!" in the middle of wifi icon), but I wasn’t able to detect any AP or Wi-Fi signal from the device. I try to erase everything and start from scratch.

grimmaldus, 2025-05-01:
@cristianchelu

So it does work, but I can only access it via the captive portal. For some reason, ESPHome OTA or Wi-Fi integration doesn’t work as expected. I’ll dig into the logs and see what might be causing the issue.

closed 2025-05-01 by grimmaldus

cristianchelu, 2025-05-01:
@grimmaldus 

That's awesome to hear!

If you want, you can submit a pull request with the `README` changes to list this as compatible, similarly to the `mb4/mb4a` row.

In that way, you'll be rightfully credited as a contributor 
and others will see right from the front door that this is compatible, instead of digging through issues.