dhewg/esphome-miot #57: Xiaomi Mi Smart Standing Fan 2 (dmaker.fan.p18) - pin layout help for flashing

smogger, 2025-05-04:
Recently disassembled this Fan and due to there is no Wiki for this specific model i will need some help in order to have a good connection to my flasher. 

This ticket was similar and talks about this model too, but the guide was a bit unclear to me. 
https://github.com/dhewg/esphome-miot/issues/19

[image]
[image]
[image]
[image]
[image]
[image]
[image]

dhewg, 2025-05-04:
According to this:
https://github.com/dhewg/esphome-miot/issues/19#issuecomment-2108367604

You'd need all of the 4 mentioned wires. I'm not sure if you can skip RTS/CTS, as I don't own that device.

But maybe @scrampker or @helgek can help out

scrampker, 2025-05-06:
If you're absolutely scared of soldering, then print up a pin holder similar to mine. I would probably make mine taller next time, to help clamp the pins harder. Pretty sure I needed all 6 wires. In theory you could program it to use the other pins that are available on the board, but honestly this is a program-once type situation. If you're good with an iron, just wire onto the ESP. Pretty straightforward.

smogger, 2025-05-09:
Since i dont have any 3d printer to create a pgo pin based jig, i was thinking to purchase an ESP-WROM for cheap, remove everything from it, and use it as a template for pogo pin solder onto it and use as a flasher jig. :D

scrampker, 2025-05-10:
I mean, you could just solder wires to the top side of the pins on the board. Should take about 5 minutes tops.

helgek, 2025-05-10:
> Since i dont have any 3d printer to create a pgo pin based jig, i was thinking to purchase an ESP-WROM for cheap, remove everything from it, and use it as a template for pogo pin solder onto it and use as a flasher jig. :D

I’ve used https://craftcloud3d.com/ several times, very useful platform.

haydenmc, 2025-05-15:
Howdy, I was able to solder directly to the pins by referencing the ESP32-WROOM-02D datasheet and flash successfully.

Here's a photo if that's helpful!

[image]

scrampker, 2025-05-16:
> Howdy, I was able to solder directly to the pins by referencing the ESP32-WROOM-02D datasheet and flash successfully.
> 
> There are big 3v3 and GND pads nearby you can use as well.
> 
> Here's a photo if that's helpful!
> 
> [image]

yep this is what I was thinking. takes a few seconds

smogger, 2025-05-28:
> Howdy, I was able to solder directly to the pins by referencing the ESP32-WROOM-02D datasheet and flash successfully.
> 
> There are big 3v3 and GND pads nearby you can use as well.
> 
> Here's a photo if that's helpful!

Sorry back from business trip and finally will have some time to disassemble again my 2 fans. 

So if i get it right>

GND / 3.3v from main board directly connected to UART in order to avoid the clutter to solder it to ESP

What i dont understand, is why do you need the ID0 pin from ESP in order to flash it? In most cases i had to work with tasmota / esp never had to use other than GND/3.3 TXD and RXD  

Thank you

haydenmc, 2025-05-28:
From memory (check datasheet) you need to hold that pin high (3.3V) when powering on the esp32 in order to allow you to re-flash new firmware.

cristianchelu, 2025-05-28:
@smogger 

> GND / 3.3v from main board directly connected to UART in order to avoid the clutter to solder it to ESP

Yes, that should spare you from having to solder those wires.

It's easier than it looks, though.

And for the IO0, I usually just insert one end of a jumper wire to a GND through-hole and press the other end against the pin with one hand while powering the board with the other (i.e. insert serial flasher in USB port), so really it's just 2 wires to actually solder.

> What i dont understand, is why do you need the ID0 pin from ESP in order to flash it? In most cases i had to work with tasmota / esp never had to use other than GND/3.3 TXD and RXD

Devkits have extra circuitry designed with frequent flashing in mind, which the bare chip inside a product doesn't need. 

Or from [Espressif themselves](https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/entering-bootloader.html):
> Espressif chips have to be reset in a certain way in order to launch the serial bootloader, only then can esptool.py communicate with the ESP chip.
> 
> On some development boards (including NodeMCU, WeMOS, HUZZAH Feather, Core Board, ESP32-WROVER-KIT), esptool can automatically trigger a reset into the serial bootloader - in which case you don’t need to read this section.

See [the automatic bootloader documentation](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#automatic-bootloader) if you're curious about more detailed info.

> esptool.py resets ESP8266 automatically by asserting DTR and RTS control lines of the USB to serial converter chip, i.e., FTDI, CP210x, or CH340x. The DTR and RTS control lines are in turn connected to GPIO0 and EN (CHIP_PU) pins of ESP8266, thus changes in the voltage levels of DTR and RTS will boot the ESP8266 into Firmware Download mode.

Have fun!

smogger, 2025-09-20:
Back to square number one. 

Got a 3rd Xiaomi Fan 2 in order to flash the ESPHome on it. 

Honestly i am starting to freak out - 3 hours of constant trying here and there but esp flashing keeps looping only under "connecting".

Tried: 
- before boot IO0 to 3V then release
- before boot IO0 to GND then release (device beeps)
- before boot IO0 to 3V keep holding
- before boot IO0 to GND keep holding
- mixing TX to RX (just for the heck)
- tried Chrome / Edge maybe it is the browser - same result
- tried flash only the "plain" flash file not the one i prebuilt - nothing
- tried different UART
- tried powering on the board from the dedicated power supply + then connecting the UART (playing with all the pinouts mentioned before)

NONE of them worked. Just to be clear i just flashed a Xiaomi Air purifier so UART / desktop "should be ok"

@scrampker  - by any chance you write down each steps you took to flash yours?
Also if you dont mind please share with me the STL file for the pinout holder. Meanwhile i bought a 3D printer so that can be my last resort if all fall...

Thank you
Some images of the struggle 

[image]
[image]
[image]
[image]
[image]
[image]

smogger, 2025-09-21:
This morning i was like ok go for direct connection. Still during flashing it stays at connecting status but nothing else. 

Checked continuity, cables/connection good. 
From the "fan lite" guide it says before connecting it, IO0 has to be connected to GND. Whenever i do i hear the beeping sound from the speaker. 

Other than these i have no clue where to go from here.

Checked fron power supply everything assembled. Fancstill working, wifi detected so i did not fried the chip. 

Any help would be highly appriciated. 

[image]
[image]

smogger, 2025-09-22:
> Do you mind updating the wiki page for this device? Like, additional help about what finally got the flashing to succeed for you (maybe with some pics from here). And it dates back to where we didn't have a commited config, it points to outdated stuff. Once that's settled, I think we can close this?
@dhewg 

Added Wiki as I was finally able to flash the 3rd fan.
https://github.com/dhewg/esphome-miot/wiki/Smart-Standing-Fan-2