dhewg/esphome-miot #70: Erreur when instal

Idaho947, 2025-06-27:
When I whan to install the yaml for the pet feeder with esphome 2025.6.1 on ha 2025.6.3, I've got some error in log :

`INFO ESPHome 2025.6.1
INFO Reading configuration /config/esphome/espgamelle.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing feeder (board: esp_wroom_02; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
INFO Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@2.0.0 has been installed!
INFO ESPAsyncTCP-esphome@2.0.0 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.3.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.3.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.3.0 has been installed!
INFO ESPAsyncWebServer-esphome@3.3.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 2MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
Compiling .pioenvs/feeder/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/event/event.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/logger/logger_esp32.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/logger/logger_esp8266.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/logger/logger_host.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/logger/logger_libretiny.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/logger/logger_rp2040.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/logger/task_log_buffer.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/miot/binary_sensor/miot_binary_sensor.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/miot/button/miot_button.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/miot/event/miot_event.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/miot/miot.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/miot/sensor/miot_sensor.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/miot/switch/miot_switch.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/miot/text_sensor/miot_text_sensor.cpp.o
In file included from src/esphome/core/log.h:12,
                 from src/esphome/components/uart/uart.h:6,
                 from src/esphome/components/miot/miot.h:10,
                 from src/esphome/components/miot/miot.cpp:4:
src/esphome/components/miot/miot.cpp: In member function 'void esphome::miot::Miot::mcu_log(const char*, ...)':
src/esphome/components/miot/miot.cpp:375:26: error: initializer fails to determine size of '__pstr__'
  375 |   ESP_LOGW(TAG, log.c_str());
/data/cache/platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
src/esphome/core/log.h:74:36: note: in expansion of macro 'F'
   74 | #define ESPHOME_LOG_FORMAT(format) F(format)
      |                                    ^
src/esphome/core/log.h:121:69: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
  121 |   ::esphome::esp_log_printf_(ESPHOME_LOG_LEVEL_WARN, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                     ^~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:154:28: note: in expansion of macro 'esph_log_w'
  154 | #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/miot/miot.cpp:375:3: note: in expansion of macro 'ESP_LOGW'
  375 |   ESP_LOGW(TAG, log.c_str());
      |   ^~~~~~~~
src/esphome/components/miot/miot.cpp:375:26: error: array must be initialized with a brace-enclosed initializer
  375 |   ESP_LOGW(TAG, log.c_str());
/data/cache/platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
src/esphome/core/log.h:74:36: note: in expansion of macro 'F'
   74 | #define ESPHOME_LOG_FORMAT(format) F(format)
      |                                    ^
src/esphome/core/log.h:121:69: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
  121 |   ::esphome::esp_log_printf_(ESPHOME_LOG_LEVEL_WARN, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                                     ^~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:154:28: note: in expansion of macro 'esph_log_w'
  154 | #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/miot/miot.cpp:375:3: note: in expansion of macro 'ESP_LOGW'
  375 |   ESP_LOGW(TAG, log.c_str());
      |   ^~~~~~~~
Compiling .pioenvs/feeder/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/feeder/src/esphome/components/number/automation.cpp.o
*** [.pioenvs/feeder/src/esphome/components/miot/miot.cpp.o] Error 1
========================= [FAILED] Took 14.04 seconds =========================
`

Can you help me ?

Idaho947, 2025-06-28:
@dhewg  Hello don't understand what I have to change. Retry today but same error again with the 6.2.

dhewg, 2025-06-28:
I commited a fix, but your local esphome caches our downloaded 
components, so you would only get the latest version after a while.

But you can add `refresh` to your config, IIRC like this:

```
external_components:
   source: ***@***.***
   refresh: 0s
```

Idaho947, 2025-06-28:
It's OK nopw thanks !