dhewg/esphome-miot #77: [E][miot:299]: Result error on property 300:300: get_down

jsapede, 2025-07-27:
i've upgraded esphome to 2025.7.3 nd now got this error in the esphome logs as well as in homeassistant logs.

here's my conig fot petfeeder, sloghly upgraded from original one with the help of christianchelu : 

```
# https://home.miot-spec.com/spec/mmgg.feeder.fi1
# 
# Home Assistant schedule card for this config available at:
# https://github.com/cristianchelu/dispenser-schedule-card

#external_components:
#  source: github://dhewg/esphome-miot@main

external_components:
  source: github://cristianchelu/esphome-miot@mmgg.feeder.fi1

substitutions:
  # Timezone for automatic daylight savings time (DST) adjustment
  # See https://en.wikipedia.org/wiki/List_of_tz_database_time_zones for options
  # Change as needed for your location
  # e.g. "Europe/London", "America/New_York", "Asia/Tokyo"
  auto_dst_timezone: "Europe/Paris" 

globals:
  - id: schedule
    # map<id, {hour, minute, portions, status}>
    type: std::map<int, std::array<int, 4>>
  - id: portions_dispensed_today
    type: int
    restore_value: True
    initial_value: "0"
  - id: auto_dst_offset
    type: int
    restore_value: True
    initial_value: "0"

script:
  - id: refresh_feed_plan
    then:
      - lambda: id(miot_main).execute_action(5, 6, "11 0");
      - lambda: id(miot_main).execute_action(5, 6, "11 1");
  - id: dispense_food
    parameters:
      portions: int
    then:
      lambda: id(miot_main).queue_command("action 2 1 5 %i", portions);
  - id: set_countrycode
    parameters:
      countrycode: int
    then:
      lambda: id(miot_main).queue_command("action 9 5 13 %i", countrycode);

esphome:
  name: distributeur-chambo
  friendly_name: Distributeur Chambo
  comment: Xiaomi Smart Pet Food Feeder (mmgg.feeder.fi1)
  project:
    name: "dhewg.esphome-miot"
    version: "mmgg.feeder.fi1"

time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: ${auto_dst_timezone}
    on_time_sync:
      lambda: |-
        id(auto_dst_offset) = id(homeassistant_time).now().timezone_offset() / 3600;
    on_time: 
      - hours: "*"
        minutes: 0
[... 392 lines cut ...]
    internal: true
    on_event:
      - event.trigger:
          id: feed_event
          event_type: "stats"
  - platform: "miot"
    miot_siid: 4
    miot_eiid: 3
    id: feedstart_event
    event_type: feedstart
    internal: true
    on_event:
      - event.trigger:
          id: feed_event
          event_type: "start"
      # Update "Dispensing" status on the UI
      - script.execute: refresh_feed_plan
  - platform: "miot"
    miot_siid: 9
    miot_eiid: 1
    id: power_loss
    name: "Power Loss"
    event_type: power_loss
    icon: mdi:power-plug-off
```

dhewg, 2025-08-02:
Unrelated to the issue at hand, but is there a reason to use 
@cristianchelu's fork? I didn't check in detail, but I hope everything 
is merged here? At least there's more fixes and updates here, which may 
or may not be relevant.

cristianchelu, 2025-08-02:
I completely overlooked that detail, great catch. 

@jsapede you should switch back to @dhewg 's repo. 

I'm no longer keeping my fork synchronized, at least until I have another reason/feature to work on, and my feeder is back on main without issues.

If I remember correctly, this is a holdover from when we tried debugging some strange behaviours, and no longer needed.