dhewg/esphome-miot #80: Clarification on Xiaomi Smart Air Purifier 4 Pro compatibility

gatesry, 2025-08-08:
Opened up a Xiaomi Smart Air Purifier 4 Pro, but I believe this uses a proprietary Xiaomi chip. It's a `MHCWB6G-B`, which appears to be close to the other non-supported devices. 

Thanks for your work on this repo!

[image]
[image]

ccandy79, 2025-08-09:
According to the FCC documentation that module contains an ESP32 C3: https://fccid.io/document.php?id=5876222

You should be able to install esphome but the zhimi.airp.mb5.yaml file won't work for you.
You will have to make your own file.

Use esp32-c3-devkitm-1 as board type.
I guess tx_pin will be GPIO7 and rx_pin will be GPIO6. But that is just a guess as I couldn't find a pinout of that module.
The spec will probably be these: https://home.miot-spec.com/spec/xiaomi.airp.vb2b

For programming use J4 as TX/RX and short J5 to enter bootloader mode.
Remember to read and save the flash before installing esphome.

gatesry, 2025-08-17:
Thanks for the confirmation! I was able to flash ESPHome onto it, but having issues with (what I believe) is the UART pin config part. As it uploads the new fw successfully, but then complains about lack of logging. 

<details>
<summary>ESPHome Firmware Build Log Snippet</summary>

```yaml
========================= [SUCCESS] Took 71.25 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.20.162 port 3232...
INFO Connected to 192.168.20.162
INFO Uploading /data/build/ryan-office-air-purifier/.pioenvs/ryan-office-air-purifier/firmware.bin (984224 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 5.07 seconds, waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
```
</details> 

Below is the API log code, highlighting it's not mapping the components correctly.  
<details>
<summary>ESPHome Firmware Build Log Snippet</summary>

```yaml
INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/office-air-purifier1.yaml...
INFO Starting log output from 192.168.20.162 using esphome API
INFO Successfully resolved office-air-purifier @ 192.168.20.162 in 0.000s
INFO Successfully connected to office-air-purifier @ 192.168.20.162 in 0.061s
INFO Successful handshake with office-air-purifier @ 192.168.20.162 in 0.062s
[12:38:45][I][app:149]: ESPHome version 2025.7.5 compiled on Aug 10 2025, 04:41:57
[12:38:45][I][app:151]: Project dhewg.esphome-miot version xiaomi.airp.va2b
[12:38:45][C][wifi:613]: WiFi:
[12:38:45][C][wifi:434]:   Local MAC: CC:4D:75:5C:17:XX
[12:38:45][C][wifi:439]:   SSID: [redacted]
[12:38:45][C][wifi:442]:   IP Address: 192.168.20.162
[12:38:45][C][wifi:446]:   BSSID: [redacted]
[12:38:45][C][wifi:446]:   Hostname: 'office-air-purifier'
[12:38:45][C][wifi:446]:   Signal strength: -51 dB ▂▄▆█
[12:38:45][C][wifi:455]:   Channel: 1
[12:38:45][C][wifi:455]:   Subnet: 255.255.255.0
[12:38:45][C][wifi:455]:   Gateway: 192.168.20.1
[12:38:45][C][wifi:455]:   DNS1: 192.168.40.10
[12:38:45][C][wifi:455]:   DNS2: 192.168.40.11
[12:38:45][C][logger:246]: Logger:
[12:38:45][C][logger:246]:   Max Level: DEBUG
[12:38:45][C][logger:246]:   Initial Level: DEBUG
[12:38:45][C][logger:252]:   Log Baud Rate: 115200
[12:38:45][C][logger:252]:   Hardware UART: USB_SERIAL_JTAG
[12:38:45][C][logger:259]:   Task Log Buffer Size: 768
[12:38:45][I][miot:215]: Network status changed to 'cloud'
[12:38:45][D][miot:195]: Queuing MCU command 'MIIO_net_change cloud'
[12:38:45][C][uart.idf:155]: UART Bus 0:
[12:38:45][C][uart.idf:156]:   TX Pin: GPIO7
[12:38:45][C][uart.idf:157]:   RX Pin: GPIO6
[12:38:45][C][uart.idf:159]:   RX Buffer Size: 256
[12:38:45][C][uart.idf:161]:   Baud Rate: 115200 baud
[12:38:45][C][uart.idf:161]:   Data Bits: 8
[12:38:45][C][uart.idf:161]:   Parity: NONE
[12:38:45][C][uart.idf:161]:   Stop bits: 1
[12:38:45][C][miot:140]: MIoT:
[12:38:45][C][miot:142]:   Model: xiaomi.airp.va2b
[12:38:45][C][miot:144]:   MCU Version: 0012
[12:38:45][C][miot:145]:   Heartbeat SIID: 0
[12:38:45][C][miot:146]:   Heartbeat PIID: 0
[12:38:45][C][miot:147]:   OTA net indicator: updating
[12:38:45][C][miot:151]:   Time: UNAVAILABLE
[12:38:45][C][miot:154]:   MCU Log:
[12:38:45][C][miot:156]:     Unknown command 'echo\x00off'
[12:38:45][C][miot:156]:     Unknown command 'ble_config\x00set 13232 0012'
[12:38:45][C][miot:156]:     Unknown command 'set_mcu_auto_ota\x00on'
[... 17 lines cut ...]
[12:38:46][W][miot:257]: Received property value without component: 2 5 2
[12:38:46][W][miot:257]: Received property value without component: 2 6 true
[12:38:46][W][miot:257]: Received property value without component: 3 1 68
[12:38:46][W][miot:257]: Received property value without component: 3 2 25.000000
[12:38:46][W][miot:257]: Received property value without component: 3 3 0
[12:38:46][W][miot:257]: Received property value without component: 3 4 1
[12:38:46][W][miot:257]: Received property value without component: 4 1 85
[12:38:46][W][miot:257]: Received property value without component: 4 2 777
[12:38:46][W][miot:257]: Received property value without component: 4 3 4223
[12:38:47][W][miot:257]: Received property value without component: 6 1 false
[12:38:47][W][miot:257]: Received property value without component: 8 1 false
[12:38:47][W][miot:257]: Received property value without component: 13 1 0
[12:38:47][W][miot:257]: Received property value without component: 14 1 11
[12:38:49][W][miot:257]: Received property value without component: 4 1 85
[12:38:49][W][miot:257]: Received property value without component: 4 2 777
[12:38:49][W][miot:257]: Received property value without component: 4 3 4223
[12:38:51][W][miot:257]: Received property value without component: 4 1 85
[12:38:51][W][miot:257]: Received property value without component: 4 2 777
[12:38:52][W][miot:257]: Received property value without component: 4 3 4223
[12:38:54][W][miot:257]: Received property value without component: 4 1 85
[12:38:54][W][miot:257]: Received property value without component: 4 2 777
[12:38:54][W][miot:257]: Received property value without component: 4 3 4223

```
</details>

ccandy79, 2025-08-17:
I'd ignore the error of the firmware update. That seems to be something like a timeout after the update. 

The UART pin config seems to be at least halfway ok as your ESP receives the properties from the device but can't understand them.
The miot2esphome utility didn't work for me (it wouldn't even compile), so I created my config manually.

Maybe you want to try and give this untested config a shot?
It's for sure not perfect but I hope it gives you some success.
The spec doesn't contain a manual fan speed control which is required by the fan component, so I took the favorite fan speed level instead.

[untested_xiaomi.airp.vb2b.yaml](https://github.com/user-attachments/files/21824286/untested_xiaomi.airp.vb2b.yaml)

gatesry, 2025-08-17:
Much appreciated - tested your config and will look to try and map the remaining missing components. Thanks so much.

closed 2025-08-17 by gatesry